esp-rs/esp-hal #3430: `esp-wifi` Linker Undefined References

theembeddedrustacean, 2025-04-27:
I'm facing an issue related to the linker, embassy, and esp-wifi. There are a few references that the linker fails to locate.

When starting with a template and modifying it to the latest dependencies, upon building, the following linker errors appear:

```bash
error: linking with `xtensa-esp32s3-elf-gcc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/Users/omarhiari/.rustup/toolchains/esp/lib/rustlib/aarch64-apple-darwin/bin:/Users/omarhiari/.codeium/windsurf/bin:/Users/omarhiari/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/local/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/bin:/var/run/com.apple.security.cryptexd/codex.system/bootstrap/usr/appleinternal/bin:/Library/TeX/texbin:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Users/omarhiari/.codeium/windsurf/bin:/Users/omarhiari/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin:/Users/omarhiari/.cargo/bin:/Users/omarhiari/Library/Application Support/Code/User/globalStorage/github.copilot-chat/debugCommand" VSLANG="1033" "xtensa-esp32s3-elf-gcc" "/var/folders/6g/wmwwyyvs30v66q68_rhfcrb80000gn/T/rustc3sfgfx/symbols.o" "<1 object files omitted>" "-Wl,--as-needed" "-Wl,-Bstatic" "/Users/omarhiari/Documents/GitHub/templatechecks3/target/xtensa-esp32s3-none-elf/release/deps/{libcompiler_builtins-11bfbbe2d487561c.rlib}" "-Wl,-Bdynamic" "-lbtbb" "-lbtdm_app" "-lcoexist" "-lcore" "-lespnow" "-lmesh" "-lnet80211" "-lphy" "-lpp" "-lsmartconfig" "-lwapi" "-lwpa_supplicant" "-lprintf" "-Wl,--eh-frame-hdr" "-Wl,-z,noexecstack" "-L" "/Users/omarhiari/Documents/GitHub/templatechecks3/target/xtensa-esp32s3-none-elf/release/build/esp-hal-d323ed4dee3455f7/out" "-L" "/Users/omarhiari/Documents/GitHub/templatechecks3/target/xtensa-esp32s3-none-elf/release/build/esp32s3-22547c892dcd6423/out" "-L" "/Users/omarhiari/Documents/GitHub/templatechecks3/target/xtensa-esp32s3-none-elf/release/build/xtensa-lx-rt-90f51266df630f25/out" "-L" "/Users/omarhiari/Documents/GitHub/templatechecks3/target/xtensa-esp32s3-none-elf/release/build/esp-wifi-sys-51664ab1c6e1817b/out" "-o" "/Users/omarhiari/Documents/GitHub/templatechecks3/target/xtensa-esp32s3-none-elf/release/deps/async_main-4346786492a2b74c" "-Wl,--gc-sections" "-no-pie" "-nodefaultlibs" "-nostartfiles" "-Tlinkall.x"
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: /Users/omarhiari/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/omarhiari/Documents/GitHub/templatechecks3/target/xtensa-esp32s3-none-elf/release/deps/async_main-4346786492a2b74c.async_main.1c78e436c1d8eb9d-cgu.0.rcgu.o:(.rwtext.literal+0x164): undefined reference to `esp_wifi_preempt_yield_task'
          /Users/omarhiari/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/omarhiari/Documents/GitHub/templatechecks3/target/xtensa-esp32s3-none-elf/release/deps/async_main-4346786492a2b74c.async_main.1c78e436c1d8eb9d-cgu.0.rcgu.o:(.literal._ZN16embassy_executor3raw20TaskStorage$LT$F$GT$4poll17hccf78389d9be8ef1E+0x25c): undefined reference to `esp_wifi_preempt_enable'
          /Users/omarhiari/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/omarhiari/Documents/GitHub/templatechecks3/target/xtensa-esp32s3-none-elf/release/deps/async_main-4346786492a2b74c.async_main.1c78e436c1d8eb9d-cgu.0.rcgu.o:(.literal._ZN16embassy_executor3raw20TaskStorage$LT$F$GT$4poll17hccf78389d9be8ef1E+0x264): undefined reference to `esp_wifi_preempt_task_create'
          /Users/omarhiari/.rustup/toolchains/esp/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/omarhiari/Documents/GitHub/templatechecks3/target/xtensa-esp32s3-none-elf/release/deps/async_main-4346786492a2b74c.async_main.1c78e436c1d8eb9d-cgu.0.rcgu.o:(.literal._ZN16embassy_executor3raw20TaskStorage$LT$F$GT$4poll17hccf78389d9be8ef1E+0x268): undefined reference to `esp_wifi_preempt_yield_task'
          collect2: error: ld returned 1 exit status
          
  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#rustc-link-lib)
```

Here are my `Cargo.toml` and  `config.toml` for reference:

```rust
[dependencies]
esp-backtrace = { version = "0.15.1", features = [
    "esp32s3",
    "exception-handler",
    "panic-handler",
    "println",
] }

esp-hal = { version = "1.0.0-beta.0", features = ["esp32s3", "unstable"] }
esp-println = { version = "0.13.1", features = ["esp32s3", "log"] }
log = { version = "0.4.21" }
esp-alloc = { version = "0.7.0" }
embedded-io = "0.6.1"

embedded-io-async = "0.6.1"
embassy-net = { version = "0.6.0", features = [
    "tcp",
    "udp",
    "dhcpv4",
    "medium-ethernet",
] }

esp-wifi = { version = "0.13.0", default-features = false, features = [
    "esp32s3",
    "wifi",
    "ble",
    "coex",
    "esp-alloc",
    "log",
] }
heapless = { version = "0.8.0", default-features = false }
smoltcp = { version = "0.12.0", default-features = false, features = [
    "medium-ethernet",
    "proto-dhcpv4",
    "proto-ipv4",
    "socket-dhcpv4",
    "socket-icmp",
    "socket-raw",
    "socket-tcp",
    "socket-udp",
] }
bleps = { git = "https://github.com/bjoernQ/bleps", package = "bleps", rev = "a5148d8ae679e021b78f53fd33afb8bb35d0b62e", features = [
    "macros",
    "async",
] }
embassy-executor = { version = "0.7.0", features = ["task-arena-size-12288"] }
embassy-time = { version = "0.4.0", features = ["generic-queue-8"] }
esp-hal-embassy = { version = "0.7.0", features = ["esp32s3"] }
static_cell = { version = "2.1.0", features = ["nightly"] }
critical-section = "1.2.0"
```

```
[target.xtensa-esp32s3-none-elf]
runner = "espflash flash --monitor"

[env]
ESP_LOG = "INFO"

[build]
rustflags = ["-C", "link-arg=-nostartfiles"]

target = "xtensa-esp32s3-none-elf"

[unstable]
build-std = ["alloc", "core"]
```

The logs are from an ESP32-S3, but I faced the same issue on an ESP32 as well.

Dominaezzz, 2025-04-27:
https://github.com/esp-rs/esp-hal/blob/main/esp-wifi/Cargo.toml#L135

bugadani, 2025-04-27:
> When starting with a template

Which template do you use that requires updating dependencies?

theembeddedrustacean, 2025-04-27:
I used `esp-generate --chip=esp32s3 [name]` 

Just looking at the repo, the template seems up to date. However, when I ran the command, I got an older template. 

I guess my pitfall was assuming that `esp-generate` updated with `cargo update` because I did run it beforehand. I just tried and noticed that I had to run the `install` command again to update and get the latest templates.

bugadani, 2025-04-27:
esp-generate bundles the template in with the binary, and is expected to be updated with/shortly after esp-hal.

closed 2025-04-29 by MabezDev