esp-rs/esp-hal #3438: xtask release preparation

MabezDev, 2025-04-28:
cc https://github.com/esp-rs/esp-hal/issues/3436 for more details

* Automatically bump dependencies, https://crates.io/crates/cargo-edit has tooling we can use to achieve this
  * Note that only we should only bump non semver major changes, major dep bumps should be done well before the release imo
  * Arguably this doesn't need to happen at release, as cargo will auto update semver compat versions, it's just nice to reflect that in the real `Cargo.toml` and not just the lock file
* A check for no path or git dependencies in the publishable crates (this is currently a manual process, should be straightforward to xtask)
* Rollover the CHANGELOG/migration files for the respective packages
  * This might be a bit tricky, but hopefully doable
* Open a PR with these changes (aka `git checkout -b branch && git commit -am "Prepare stage1 release" && git push -u` + some stuff for opening a PR via github, not sure of the tooling available here)
* For now the bumping the _package versions_ themselves we'll do manually
  * Either via the web editor (press `.` in the PR) or a local fixup. 
  * We can improve this in the future, but requires too much work to think about doing now

bugadani, 2025-04-29:
> Automatically bump dependencies

That should be a job for renovate or similar dependency management bots.

bjoernQ, 2025-04-29:
Maybe the nightly-CI could also tell us about outdated dependencies? But not in a too annoying way since sometimes we need/want to stay on older version 🤔

PR #3542 (2025-05-26): Publish based on release plan
Closes #3438
Closes #3437
Closes #3436
closed 2025-05-26 by bugadani