esp-rs/esp-hal #3450: ESP32 Incomplete UART Write String Output

theembeddedrustacean, 2025-05-04:
On the ESP32, switching between `esp-hal` `0.23.1` and `1.0.0-beta 0` results in different UART output behaviour when printing long strings. In the latter, the output does not print out fully.

Here's the code for the example output:

```rust
#[main]
fn main() -> ! {
    let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let peripherals = esp_hal::init(config);

    let mut serial = {
        let (tx_pin, rx_pin) = (peripherals.GPIO1, peripherals.GPIO3);
        let config = Config::default();
        Uart::new(peripherals.UART0, config)
            .unwrap()
            .with_tx(tx_pin)
            .with_rx(rx_pin)
    };

    serial
        .write_str(
            "This is a very long string that will be printed to the serial console.
First line of the string.
Second line of the string.
Third line of the string.
Fourth line of the string.
Fifth line of the string.
Sixth line of the string.",
        )
        .unwrap();

    loop {}
}
```

Output when using `0.23.1`:
[image]

Output when using `1.0.0-beta 0`:
[image]

theembeddedrustacean, 2025-05-04:
On a side note, the replacement characters potentially resulting from transfer errors don't appear immediately. It takes them a while, after a few seconds or more in some cases, when the transfer is supposed to be completed.

closed 2025-05-05 by JurajSadel

Frostie314159, 2025-05-22:
This also applies to the `core::fmt::Write` implementation for UART, which also truncates the output.

MabezDev, 2025-05-22:
The PR also tackled this: https://github.com/esp-rs/esp-hal/pull/3452/files#diff-44c26a357c3370ef8d7c4748ff6930fd0b4c6f75d705ce823440a42687c358d1R1768 are you still running into this?

Frostie314159, 2025-05-22:
Oh, I'm still on the crates.io release. Thanks for pointing that out.