esp-rs/esp-hal #3454: Is there any plan to add ESP32-C5 support?

windoze, 2025-05-06:
## Motivations

I saw [ESP32-C5 DevKeyC-1 is available](https://item.taobao.com/item.htm?abbucket=18&detail_redpacket_pop=true&id=846843964795&mi_id=CmK-AWvyRoejOOdxkkQpNlNjmv41i_iEkJWwtL9HmD04aW9f9DT7ew29opPHYnR4IqM0GvxqCvpmn2nAWcVZ7MWJxEgwoysnfREgGUJkB-g&ns=1&priceTId=214781b517465098897396881e85e9&query=esp32-c5%20devkitc%201&skuId=5787598760962&spm=a21n57.1.hoverItem.1&utparam=%7B%22aplus_abtest%22%3A%22fe5cbf3accb638b37ba67d8ba1cd9333%22%7D&xxc=taobaoSearch) now, is there any plan to add support?
According to [the doc](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c5/esp32-c5-devkitc-1/index.html), it has dual band Wi-Fi, which makes it a better alternative for IoT projects that need higher network bandwidth.

jessebraham, 2025-05-06:
Yes, we do plan on adding support for this device eventually. Unfortunately, we are unable to give a timeline at this point. I asked for a devkit yesterday and am unable to obtain one at this time for whatever reason. Either myself or another maintainer will share any updates we have on this situation as soon as we're able.

windoze, 2025-05-17:
[image]
Just got a ESP32-C5-DevKitC-1, is there any thing I can do?

maxtnuk, 2025-05-19:
for supporting this, we shoul implement this crate first
https://github.com/esp-rs/esp-pacs

bugadani, 2025-05-19:
Thank you all, the C5 will be supported as soon as we can add support for it.