esp-rs/esp-hal #3473: rustfmt: Consider enabling `error_on_line_overflow`

bjoernQ, 2025-05-08:
Without enabling `error_on_line_overflow` rustfmt will just not touch lines where the width exceeds `max_width` (=100) resulting in strange formatting

Enabling it reveals _a lot_ of places where this currently is the case

bjoernQ, 2025-05-19:
I guess we could just do it - the xtask uses nightly for the fmt checks already (because we are using some other nightly-onlies)

playfulFence, 2025-09-05:
After some analysis (thanks @jessebraham for an insight too!), I've noticed, that every instance of `max_width` being exceeded is somehow related to macro: either is a macro definition or the code is under a macro. And then I found [this discussion](https://github.com/rust-lang/rustfmt/discussions/5437), which clearly explains why that happens, but TL;DR : rustfmt works on the AST, but macros are seen as raw tokens before expansion — so if the tokens don’t form valid Rust syntax, rustfmt doesn’t know how to format them.

Manually editing/formatting every case is of course possible, but may lead into even less readable and uglier code. IMO, what we should do is somehow try to minimize macro usage in our codebase or wait and hope that `rustfmt` will one day learn how to work with them 😅 .

But as for this particular problem, maybe it is better to leave everything as it is than to try to “fix” all 100+ cases of this kind (screenshot)

[image]

bjoernQ, 2025-09-05:
the whole reason for considering this would be getting rustfmt to do something with the macro invocation - but currently this is somewhat limited: https://github.com/rust-lang/rustfmt/discussions/5437#discussioncomment-3117043

Especially since it ignores macro-invocations using brace delimiters `{ ... }` changes this to a chore without much to win

So probably we can abandon this idea for now

closed 2025-09-05 by playfulFence