esp-rs/esp-hal #3488: esp32c3 does not connect to Wifi

oleid, 2025-05-10:
## Bug description

It would appear that my esp32c3 cannot connect to my home wifi, while the original esp32 connects without any issues.

## To Reproduce

This is what I did:

1. Check-out `main` branch of `esp-hal` repo
1. Update the compiler (`espup update ; rustup update`)
1. Export wifi settings as environment variables.
2. I added an assert and the wifi name to the example debug output to verify if the settings are correct.
3. Run: `cargo xtask run example examples esp32c3 --example wifi_embassy_dhcp`

```
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.06s
     Running `target/debug/xtask run example examples esp32c3 --example wifi_embassy_dhcp`
[2025-05-10T11:48:52Z INFO  xtask] Building example 'src/bin/wifi_embassy_dhcp.rs' for 'esp32c3'
[2025-05-10T11:48:52Z INFO  xtask]   Features:      embassy, esp-hal/unstable, esp-wifi, esp-wifi/wifi
    Updating crates.io index
    Finished `release` profile [optimized + debuginfo] target(s) in 5.09s
     Running `espflash flash --monitor target/riscv32imc-unknown-none-elf/release/wifi_embassy_dhcp`
[2025-05-10T11:48:57Z INFO ] Detected 4 serial ports
[2025-05-10T11:48:57Z INFO ] Ports which match a known common dev board are highlighted
[2025-05-10T11:48:57Z INFO ] Please select a port
✔ Remember this serial port for future use? · no
[2025-05-10T11:49:01Z INFO ] Serial port: '/dev/cu.usbmodem2101'
[2025-05-10T11:49:01Z INFO ] Connecting...
[2025-05-10T11:49:02Z INFO ] Using flash stub
Chip type:         esp32c3 (revision v0.3)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       60:55:f9:28:2a:74
App/part. size:    665,472/4,128,768 bytes, 16.12%
[2025-05-10T11:49:02Z INFO ] Segment at address '0x0' has not changed, skipping write
[2025-05-10T11:49:02Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2025-05-10T11:49:02Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2025-05-10T11:49:03Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380562
0x40380562 - core::num::<impl u32>::trailing_zeros
    at /Users/oleid/.rustup/toolchains/stable-aarch64-apple-darwin/lib/rustlib/src/rust/library/core/src/num/uint_macros.rs:162
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x1714
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f9c
entry 0x403cc710
I (24) boot: ESP-IDF v5.1.2-342-gbcf1645e44 2nd stage bootloader
I (24) boot: compile time Dec 12 2023 10:50:58
I (25) boot: chip revision: v0.3
I (29) boot.esp32c3: SPI Speed      : 40MHz
I (34) boot.esp32c3: SPI Mode       : DIO
I (38) boot.esp32c3: SPI Flash Size : 4MB
I (43) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 003f0000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=3c000020 size=1555ch ( 87388) map
I (114) esp_image: segment 1: paddr=00025584 vaddr=3fc87ba8 size=01ee8h (  7912) load
[... 167 lines cut ...]

- target device:

Working:

```
Chip type:         esp32 (revision v1.0)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       24:6f:28:7a:00:08
```

Not working:

```
[2025-05-10T11:54:31Z INFO ] Using flash stub
Chip type:         esp32c3 (revision v0.3)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       60:55:f9:28:2a:74
```

- Crate name and version: esp-hal  @ 7b7844a85516d0a5c00df6b0e5f5a251abf323bb

giomf, 2025-05-15:
What esp32c3 board are you using? It is able to connect to other wifis?

I had the same issue using a esp32c3 board from Seedstudio without connecting the antenna via u.FL.

oleid, 2025-05-15:
Hi @giomf, thanks for your input.

> What esp32c3 board are you using
> I had the same issue using a esp32c3 board from Seedstudio without connecting the antenna via u.FL.

I honestly don't recall where I bought it, since I've in in my drawer for some time now. But here are a few photos.

[image]
[image]

> Your signal strength is at -83 which is quite low if your AP is in the same room as your esp board.

Indeed, but mind that value was measured by the older esp32. Possibly due to using a mesh wifi. Might be the other access point. Wifi-Analyzer yields -56dBm on my phone.

I didn't expect wifi strength to be a problem, though. I'll try connecting when the device is closer to the access point. 
They are in the same room, but at opposite ends.

> Do you have the option to add an external antenna?

Not sure, I'll have to research this.

giomf, 2025-05-15:
Ok, your board has an onboard antenna and I dont see any connector for an external antenna. However,  I think you onboard antenna should be enough to connect to your wifi.

Token-Thinker, 2025-05-17:
> What esp32c3 board are you using? It is able to connect to other wifis? Your signal strength is at -83 which is quite low if your AP is in the same room as your esp board.
> 
> I had the same issue using a esp32c3 board from Seedstudio without connecting the antenna via u.FL. Do you have the option to add an external antenna?

Can confirm that fix worked...........2 days later -.- 

[image]

oleid, 2025-05-17:
I now moved the device  into an  `~ 1 m` radius to the device.
The scan now has results.

````
cargo xtask run example examples esp32c3 --example wifi_embassy_dhcp
[...]
I (86) esp_image: segment 0: paddr=00010020 vaddr=3c000020 size=15594h ( 87444) map
I (114) esp_image: segment 1: paddr=000255bc vaddr=3fc87ba8 size=01ef0h (  7920) load
I (116) esp_image: segment 2: paddr=000274b4 vaddr=3fca48c0 size=001ech (   492) load
I (120) esp_image: segment 3: paddr=000276a8 vaddr=40380000 size=07ba8h ( 31656) load
I (136) esp_image: segment 4: paddr=0002f258 vaddr=00000000 size=00dc0h (  3520) 
I (137) esp_image: segment 5: paddr=00030020 vaddr=42020020 size=82790h (534416) map
I (266) boot: Loaded app from partition at offset 0x10000
I (266) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration EspWifiConfig { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: true, ampdu_tx_enable: true, amsdu_tx_enable: false, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
WARN - esp_wifi_internal_tx 12290
start connection task
Device capabilities: Ok(EnumSet(Client | AccessPoint | Mixed))
Starting wifi
Wifi started!
Scan
AccessPointInfo { ssid: "Katzenparadies", bssid: [60, 124, 63, 84, 22, 89], channel: 1, secondary_channel: None, signal_strength: -50, auth_method: Some(WPA2Personal) }
AccessPointInfo { ssid: "Katzenfalle", bssid: [60, 124, 63, 84, 22, 90], channel: 1, secondary_channel: None, signal_strength: -50, auth_method: Some(WPA2Personal) }
AccessPointInfo { ssid: "AP_2795277764", bssid: [68, 42, 143, 6, 56, 254], channel: 9, secondary_channel: None, signal_strength: -70, auth_method: Some(WPA2Personal) }
About to connect to Katzenfalle with PW XXXXXXXX...
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
Failed to connect to wifi: Disconnected
About to connect to Katzenfalle with PW XXXXXXXX...
WARN - esp_wifi_internal_tx 12294
Failed to connect to wifi: Disconnected
About to connect to Katzenfalle with PW XXXXXXXX...
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
Failed to connect to wifi: Disconnected
WARN - esp_wifi_internal_tx 12294
About to connect to Katzenfalle with PW XXXXXXXX...
Failed to connect to wifi: Disconnected
WARN - esp_wifi_internal_tx 12294
About to connect to Katzenfalle with PW XXXXXXXX...
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
Failed to connect to wifi: Disconnected
About to connect to Katzenfalle with PW XXXXXXXX...
Failed to connect to wifi: Disconnected
WARN - esp_wifi_internal_tx 12294
About to connect to Katzenfalle with PW XXXXXXXX...
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
Failed to connect to wifi: Disconnected
WARN - esp_wifi_internal_tx 12294
About to connect to Katzenfalle with PW XXXXXXXX...
Failed to connect to wifi: Disconnected
About to connect to Katzenfalle with PW XXXXXXXX...
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
WARN - Attempt to free null pointer
````

(Replaced the real password with `XXXXXXXX`)

With

`git diff`

```diff
diff --git a/examples/src/bin/wifi_embassy_dhcp.rs b/examples/src/bin/wifi_embassy_dhcp.rs
index 9030e02d..cc6ac330 100644
--- a/examples/src/bin/wifi_embassy_dhcp.rs
+++ b/examples/src/bin/wifi_embassy_dhcp.rs
@@ -178,7 +178,7 @@ async fn connection(mut controller: WifiController<'static>) {
                 println!("{:?}", ap);
             }
         }
-        println!("About to connect...");
+        println!("About to connect to {SSID} with PW {PASSWORD}...");
 
         match controller.connect_async().await {
             Ok(_) => println!("Wifi connected!"),
```

So according to the scan, I think the Wifi signal should be strong enough. Yet, the connection is not successful :/

giomf, 2025-05-22:
You could try to get the logs from the access point and also run your program on the esp with the debug log level to get more information. At least the access point should log why it will not accept the connection.

Lenoirio, 2025-05-22:
I've had a lot of issue with those supermini S3 (C3 also know for that) modules that you can get for a few $$$ on e.g. AliX.
According to some info I could get they have some bad antenna design (too close to the chip, ignoring the recommendations of the chip manufacturer). Also soldering an external antenna did not help.

I nearly gave up but then I found a discussion on YouTube where someone mentioned that *lowering* the TX-power solved the issue. And it also solved the issue for me. Although having excellent RSSI value when listing the APs, I was not able to get a working DHCP configure.

I added this code and all of a sudden it worked:
 // This is needed for the ESP32-S3 supermini (max=80 -> 20dbm; it seems that *lowering* the power solves a connection issue)

unsafe {
        let r = esp_wifi_sys::include::esp_wifi_set_max_tx_power(8);
        log::info!("Set max WiFi power result {:?}", r);
}

The explanation in the discussion was that the ESP has some self-protection if TX is "backfiring" into the chip and so stops transmitting.
I can't promise anything but worth a try.