esp-rs/esp-hal #3491: Make esp_wifi::deinit() return consumed resources

sssilver, 2025-05-15:
## Motivations

Presently, the `esp_wifi::init(tg, rng, radio_clk)` function consumes its input resources — specifically the timer group, random number generator, and radio clocks — to initialize the BLE module. This, combined with the fact that `esp_wifi::deinit()` does not return those resources back, makes it difficult to dynamically power BLE off and on at runtime.

If you want to disable BLE and later re-enable it, you would need to reacquire those same hardware peripherals. However, these are typically only accessible once at boot via `peripherals::<...>`, making reacquisition only practically possible in `main()`.

This severely limits flexibility for applications that need to control BLE power dynamically in response to runtime conditions like battery state, user commands, or peripheral conflicts.

## Solution

Have `esp_wifi::deinit()` return the consumed hardware resources (e.g., `Timer`, `Rng`, and `RADIO_CLK`) back to the caller. This would allow them to be reused later in another call to `init()`, enabling clean BLE reinitialization during runtime without resorting to unsafe peripheral access or full system reset.

## Alternatives

* Implementing a manual peripheral recovery layer via `unsafe` access to `Peripherals` — fragile and error-prone.
* Avoiding BLE power cycling entirely — wasteful for power-sensitive applications.
* Performing a full chip reset — impractical for many embedded scenarios.

## Additional context

This change would make it significantly easier to model BLE as a dynamic state machine in higher-level embedded abstractions, allowing for safe and idiomatic Rust code that manages power without leaking or duplicating hardware ownership.

MabezDev, 2025-05-19:
What you're asking for is currently possible, just in a different way. You can pass the resources in via mutable refererence (if you're using main, you'll need to use the `.reborrow()` method). If you do this the driver only "borrows" the resources whilst the driver is up, and once `deinit` is called the borrow is "dropped" and you can use the resources again however you like.

Here is a small code snippet to nudge you in the right direction:

```rust
let controller = esp_wifi::init(&mut tg, /* other stuff */);
// info!("tg: {}", tg); // compiler error as tg is still in use
drop(controller)
// info!("tg: {}", tg); // fine as tg is now not borrowed by anything
```

How can we explain this better? It's something we've struggled to make "discoverable", do you have any ideas?

sssilver, 2025-05-20:
Fascinating -- thank you for clarifying this.

Just a small follow-up -- what do you mean by:

> if you're using main, you'll need to use the `.reborrow()` method

A simple way to boost discoverability is to document this problem and its solution in a single, well-crafted paragraph in `esp_wifi::init()`’s docstring, which I do remember reading when I was exploring this.

A better way that'd take more work is to publish one markdown-formatted page that explains how to work with the Wifi/BLE stack, common scenarios and solutions to those scenarios.

bugadani, 2025-05-20:
Please read through [this](https://github.com/esp-rs/esp-hal/blob/aeaab9a95b7e961e31a4becc19f4b6970cdb1af6/esp-hal/src/lib.rs#L31-L75) and let me know if it needs to be changed in some way.

sssilver, 2025-05-20:
@bugadani the problem with what you've linked is that it only makes sense after reading the comment of @MabezDev in reference to the problem I was having.

If I'm at the point in my project development where I have decided I need to power off and on my BLE module, it's incredibly unlikely that I will go and start reading the docstring in lib.rs of esp-hal itself.

Moreover -- even if what's written there fully makes sense after @MabezDev's comment, I'm not sure I would even connect the dots if I had discovered it on my own (which I wouldn't have), given that the example only mentions I2C and says nothing about BLE.

In the real world, overwhelming majority of software engineers don't read docs until they are stuck. And when they _are_ stuck, they generally start reading information available to them near the point where they get stuck. In this real-world paradigm, then, a docstring in the lib.rs is pretty far away from that point.

When I wanted to power off and on my BLE device, the first thing I tried was typing `peripheral.` in my IDE to see if there's any function in that list that sounds like it may be related to powering the peripheral off and on.

[image]

There wasn't. Then I googled something like "esp ble power off and on rust". There was nothing. Then I asked ChatGPT, Claude, and Gemini. They knew nothing.

I went back to my IDE and got all the way back to `init`, then found `deinit()` in it, then started struggling with the type system, asked questions in the ESP32 matrix chatroom, and eventually created this issue. Clearly, my problem wasn't laziness -- I spent days working hard on this -- I simply wasn't intelligent enough.

Ideally, the library/API would have made this apparent to me during one of the earlier steps of the journey I described above. Note that in my process I never thought about reading lib.rs. I never even thought about googling "esp32 peripheral singletons power consumption reborrow", which -- ironically -- doesn't even find the lib.rs paragraph you've linked.

MabezDev, 2025-05-20:
Thanks for the feedback!

I think the takeaway here is we should link back to certain resources in applicable places. For instance, we can add a link back to the esp-hal concept section in the docs for the peripheral singletons, and also we may wish to add a line to certain constructors (and esp_wifi::init).

We'll have a book section on this at some point too.

> Then I googled something like "esp ble power off and on rust"

We could also improve our examples here, we have a few issues about this already, but this is something we can demonstrate in a more complete example at some point.