esp-rs/esp-hal #3497: Requesting practical examples

David-OConnor, 2025-05-17:
## Motivations

When someone is new to the library, or new to the newest revisions, it would be helpful to have practical example code to reference. Everyone's program is different, and some peripherals and peripheral functionality is more common than others, so there will not be a perfect solution, but I believe we can augment the ones present in the `examples` folder.

## Solution

We should populate the examples folder with the following. These can be combined into a smaller number of files (Or ideally, full standalone project examples) They should demonstrate the following as a basline:

- Timer init, config, and interrupts.
- UART init, config, interrupts, blocking and DMA reads and writes.
- SPI init, config, interrupts, blocking and DMA reads and writes. (Slave, at least)
- I2C init, config, interrupts, blocking and DMA reads and writes. (Slave, at least)
- Using shared state, e.g. that is available in one or more interrupt contexts. This should be integrated into each of the other examples; i.e, not a new "shared state" example.

Note that there is an inherent friction between the current example setup of individual, feature-specific files that allows conveniences like `cargo xtask run-example esp-hal esp32c6 hello_world`, and having full, standalone example folders you can clone and run. I don't offer a solution for this, other than perhaps include a few multi-file examples that don't work with that system, or to use single-files that have the `module{}` syntax used, where in a practical project they might be separate files. 

## Alternatives

Instead of populating the examples folder, alternatives include detailed doc comments with examples for use on Rust Docs, moving to a less trait-heavy approach so that rust docs are easy to follow from structs, enums, and their methods and fields, and using a dedicated documentation website of a suitable form.

I think that populating the examples folder on Github would be easier than any of the above, but they would also solve this problem.

## Additional context

I am an experienced rust and embedded developer, and am having a hard time making a basic program using the current (e.g. from Git; soon to be released as 1.0) ESP-HAL. I think something like [these examples](https://github.com/David-OConnor/stm32-hal/tree/main/examples) I set up in STM32-HAL, and [this practical quickstart](https://github.com/David-OConnor/stm32-hal-quickstart) demonstrate the form that would help, but using ESP-HAL's syntax.

I consider the tasks I don't see documented as fundamental to most embedded programs. I think it might help to understand this mindset by taking a step back, and asking yourself "If I had never seen this library before, could I figure out how to make a simple (practical) program using the information published?"

Thank you for all the great work towards 1.0.0, and making Espressif microcontroller use on Rust a reality! This is an important capability for using rust on embedded due to Espressif's, and esp-HAL and related ecosystem's support of Wi-Fi and BLE. This issue's purpose is to enable people to be able to take advantage of this work. I know the functionality is there; let's see it! :)

Dominaezzz, 2025-05-17:
https://github.com/esp-rs/esp-hal/issues/2596

bugadani, 2025-05-19:
> I2C init, config, interrupts, blocking and DMA reads and writes. (Host, at least)

Just FYI our I2C hardware is not DMA-capable.

bugadani, 2025-05-19:
> I think that populating the examples folder on Github would be easier than any of the above, but they would also solve this problem.

We used to have multiple dozens of examples. They are impossible to maintain: when you need to update them for API changes, they quickly become a time sink that just hinders development and blow up PR sizes for no good reason.

We have https://github.com/esp-rs/awesome-esp-rust that link to nontrivial projects. We can also have *external* examples that we (or the community) would/could update after esp-hal releases. Updating examples once instead of any number of times in a development cycle is a lot more appealing approach.

Also note that a lot of what you ask for is unstable. In the case of interrupts and DMA maintaining examples don't make much sense - not only we'd have to make a lot of them and keep them up to date, but the APIs themselves are likely vastly different now than what they will be stabilized as. Once we stabilize an API, providing an example with it becomes much less of a pain point.

David-OConnor, 2025-05-19:
I will simply say, I disagree. See the demonstration links I posted. I made API changes last week. Went through the examples folder changing syntax; it took 10 minutes of copy+pasting. The current alternative I observe is an undocumented API.

Dominaezzz, 2025-05-19:
> The current alternative I observe is an undocumented API.

Which APIs are undocumented? 

I think you'll have a better chance asking for documentation than more examples.

The current thinking is the documentation should be enough to get you going rather than examples.

If you really do just want full examples then this issue can be closed as a duplicate of the issue I posted earlier.

David-OConnor, 2025-05-19:
I agree that *Discovery* is a good characterization of this; great call. 

I think that the book would be a suitable alternative; maybe better than examples. I disagree that examples couldn't also solve it. I suggested that as the primary approach because it's an easy and fast approach that would meet the intent, while the book would be more involved due to mixing narratives and code samples in a coherent way. Either would meet the intent, IMO.

MabezDev, 2025-07-18:
Thanks for the valuable input! 

Whilst I can appreciate good documentation is in the eyes of the beholder, and everyone wants something slightly different, I do believe for our stable drivers at least we have good documentation that outlines all the things in your bullet points, regarding init and config. DMA is a whole other avenue of work to be explored before we can settle on stabilization and _no_ HAL has a perfect DMA story yet.

> I agree that Discovery is a good characterization of this; great call.
> 
> I think that the book would be a suitable alternative...

We're in the midst of a book rewrite, which might help explain these concepts a bit better too, see https://github.com/esp-rs/book/milestone/2.

I've opened https://github.com/esp-rs/esp-hal/issues/3830 to put us on the path of https://github.com/esp-rs/esp-hal/issues/2596. The examples likely won't be "less trivial", at least at the start but switching to project based examples gives us the ability to in the future.

Thanks again, closing for now.

closed 2025-07-18 by MabezDev

David-OConnor, 2025-07-18:
>  DMA is a whole other avenue of work to be explored before we can settle on stabilization and no HAL has a perfect DMA story yet.

Perfect isn't the goal for any of this. The HAL I wrote for STM32 has good DMA support, and docs for it, and it's not the only one.