esp-rs/esp-hal #3498: Flex::apply_output_config does not init output

abreis, 2025-05-18:
I found this behaviour a bit unexpected. The following setup:
```rust
let mut pin = gpio::Flex::new(some_pin);
let output_config = gpio::OutputConfig::default()
    .with_drive_mode(gpio::DriveMode::OpenDrain)
    .with_pull(gpio::Pull::None)
    .with_drive_strength(gpio::DriveStrength::_40mA)
pin.apply_output_config(&output_config);
pin.set_high();
pin.enable_input(true);
```
does not seem to enable the pin's output unless this call is also added:
```rust
pin.set_as_open_drain(gpio::Pull::None);
```
which is also somewhat redundant as both `DriveMode` and `Pull` were just set by `OutputConfig`.

Building the `Flex` pin the other way around works correctly:
```rust
let mut pin = gpio::Output::new(
    some_pin,
    gpio::Level::High,
    gpio::OutputConfig::default()
        .with_drive_mode(gpio::DriveMode::OpenDrain)
        .with_pull(gpio::Pull::None)
        .with_drive_strength(gpio::DriveStrength::_40mA),
)
.into_flex();
pin.enable_input(true);
```

I think that: 
- setting an `OutputConfig` on `Flex` should enable the output, or
- the documentation for `Flex::apply_output_config` should mention that it does not enable the output.

bugadani, 2025-05-18:
This went through a bit of a revision in https://github.com/esp-rs/esp-hal/pull/3387, hopefully we can release this soon enough.

abreis, 2025-05-18:
Ah, apologies, I didn't see that PR. The new interface seems much better.

closed 2025-05-18 by abreis