esp-rs/esp-hal #3508: embassy-usb-driver for esp32c6 (UsbSerialJtag)

MonaMayrhofer, 2025-05-21:
## Motivations
I am working with postcard and postcard-rpc, which has pre-made server implementations for embassy-usb, however as far as I can see, the relevant embassy-usb-driver traits are only implemented for the USB-OTG peripheral on a few esp32s* chips. 

## Solution
I would love to have the embassy-usb-driver traits implemented for the UsbSerialJtag (probably when the DriverMode is set to Async)

## Alternatives
A workaround is that users of postcard-rpc would implement their own WireRx and WireTx traits ontop of the UsbSerialJtag interface, however that only solves that one particular application, and from what I - with my limited esp-hal codebase knowledge - can tell, it should be more generally useful to integrate with the embassy-usb-driver traits.

## Additional context
If there isn't some fundamental reason why the embassy-usb-driver traits cannot be implemented for the esp32c6 that I have not understood yet, I'd be willing to put in some work for a PR, however I am rather new to the embedded rust ecosystem so I would likely need some help and pointers into the right directions of what exactly would need to be done for that!

bugadani, 2025-05-21:
Unfortunately what you ask for is not possible to do. The USB_SERIAL_JTAG peripheral is not a generic-purpose USB device, its endpoints and interfaces are fixed. Changing this isn't a matter of implementation effort.

I do recommend you taking a look at [this PR](https://github.com/jamesmunns/postcard-rpc/pull/96) that demonstrates how you can use the USB_SERIAL_JTAG peripheral with postcard-rpc. Note that this is blocked on https://github.com/jamesmunns/postcard-rpc/pull/83 - I think we may need to politely ping @jamesmunns that there is more interest in getting that particular questline completed, and I'm sure he'll let us know how we can help :)

closed 2025-05-21 by bugadani

jamesmunns, 2025-05-21:
Just need more time in the day, unfortunately. Please do keep repinging! One of these days I'll have time...