esp-rs/esp-hal #3529: RFC: Use `libesp_rom.a` for ESP32 in esp-storage

bjoernQ, 2025-05-22:
In esp-storage we cannot just use the ROM functions like we do for all other chips - we need "patched" versions of the ROM functions.

Instead of reproducing that code we could just use `libesp_rom.a` - while it's a blob the can reference the source (which we cannot do for the actual functions residing in ROM)

Then the ESP32 code would look like what we have for all other chips

Easiest would be to add the lib in esp-storage but we could also do that in esp-hal, latest when we implement #3389 that would be a good place for this

MabezDev, 2025-05-23:
Slightly adjacent, but very related: How hard would this be to _not_ use ROM code at all on any of the chips?

I like this ideal in general though, and given the ROM code works well on the other chips it would be nice to have something that "looked" similar.

igrr, 2025-05-23:
Just a couple of things to add:

1. On newer chips (C3/S3 and later) there is a new [esp_flash driver](https://github.com/espressif/esp-idf/blob/master/components/spi_flash/include/esp_flash.h) in ROM which can be used instead of the legacy SPI flash functions. It has the advantage that if you ever need to add a driver for a new SPI flash vendor, you can do so without having to reimplement or patch all of the driver functions.
2. Sometimes the ROM driver is missing a few bugfixes (e.g.: no support for addressing >16 MB flash in P4 ROM), those have to patched in later anyway.

bjoernQ, 2025-07-02:
Closed via #3688