esp-rs/esp-hal #3556: flip-link allways overflows

PaulekOfficial, 2025-05-27:
## Bug description

When enabling the future ESP_HAL_CONFIG_FLIP_LINK in config.toml region 'RAM' always overflows.

ctions" "-Map=target/memory.map" "--nmagic" "-Tdefmt.x" "-Tlinkall.x"
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: rust-lld: error: section '.bss' will not fit in region 'RAM': overflowed by 352 bytes
          rust-lld: error: section '.noinit' will not fit in region 'RAM': overflowed by 352 bytes
          rust-lld: error: section '.data.wifi' will not fit in region 'RAM': overflowed by 352 bytes

## To Reproduce

Clone my repository go to .cargo/config.toml uncomment the 8th line with #ESP_HAL_CONFIG_FLIP_LINK = "true" and build this project.

https://github.com/PaulekOfficial/bg95-cmux-esp32c6/tree/main

## Expected behavior

Project should build without an issue.

## Environment

- Target device: ESP32C6 using probe-rs
- Crate name and version: esp-hal 1.0.0-beta.0

PaulekOfficial, 2025-05-27:
Changing the order of linking scripts gives this same error. 

rustflags = [
  # Required to obtain backtraces (e.g. when using the "esp-backtrace" crate.)
  # NOTE: May negatively impact performance of produced code
  "-C", "force-frame-pointers",

  "-C", "link-args=-Map=target/memory.map",
  "-C", "link-args=--nmagic",
  "-C", "link-arg=-Tdefmt.x",
  "-C", "link-arg=-Tlinkall.x",
]

or

rustflags = [
  # Required to obtain backtraces (e.g. when using the "esp-backtrace" crate.)
  # NOTE: May negatively impact performance of produced code
  "-C", "force-frame-pointers",

  "-C", "link-arg=-Tlinkall.x",
  "-C", "link-args=-Map=target/memory.map",
  "-C", "link-args=--nmagic",
  "-C", "link-arg=-Tdefmt.x",
]

still causes

  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: rust-lld: error: section '.bss' will not fit in region 'RAM': overflowed by 352 bytes
          rust-lld: error: section '.noinit' will not fit in region 'RAM': overflowed by 352 bytes
          rust-lld: error: section '.data.wifi' will not fit in region 'RAM': overflowed by 352 bytes

bjoernQ, 2025-05-27:
I think it's (at least partially) caused by this section
```
.uninit.defmt-rtt.BUFFER 40800db4    1024 RAM      [▏                                                                     ]
```

We should add `.uninit.*` to https://github.com/esp-rs/esp-hal/blob/631b4aab410cfd1aabe101f2b883101635668e77/esp-hal/ld/sections/rwdata.x#L50