esp-rs/esp-hal #3583: Revisit ESP32-S2 SVD's SPI0 / SPI1 definitions

bjoernQ, 2025-06-02:
The SVD treats all SPIs the same while in reality SPI0/SPI1 are MEM-SPI and others are GP-SPI. They have some things in common but also some differences in register layouts

e.g. `ctrl::fcmd_quad` is defined as bit 9 (correct for GP-SPI) while it's bit 8 for MEM-SPI - a real-world problem can be seen in #3554 

(Needs to get fixed in esp-pacs but we also need to make sure things still work (that's why the issue is here) - also I guess we don't want to YOLO-fix it before beta.1)

playfulFence, 2025-07-11:
Now, as https://github.com/esp-rs/esp-pacs/issues/335 is resolved, we can do all the necessary changes when we'll be updating used PAC version
https://github.com/esp-rs/esp-hal/blob/main/esp-hal/src/soc/esp32s2/psram.rs#L486

bugadani, 2025-07-17:
Next time maybe we should fix these things before we release esp-hal.

closed 2025-07-17 by MabezDev