esp-rs/esp-hal #3638: Reparate `esp-riscv-rt` from `esp-hal`

gibbz00, 2025-06-14:
Hey again,

As discussed in the matrix chat [^0], I'm investigating the possibility of my ESP32C3 without needing to pull `esp-hal`, but by only using the `rt`.

One major blocker for this seems to be how `esp-riscv-rt` takes a roundtrip to `esp_hal::hal_main` before executing `main` [^1]. From what I can tell, its function seems to be to setup the experimental nightly stack smashing protection feature as introduced here: [^2]. (Third paragraph in [^3] is quite an interesting read.) Anyhow, I was wondering if you (the maintainers?) would be open to adding a `no-hal` feature flag in `esp-riscv-rt` which replaces the required roundtrip [^4] to:

```rust
#[cfg(not(feature = "no-hal"))]
hal_main(a0, a1, a2);
#[cfg(feature = "no-hal")]
main(a0, a1, a2);
```

I've honestly not yet taken a look at how the linker scripts and symbol definitions in the hal crates could be reused when enabling the `no-hal` feature. Suppose that would be the next step in coming up with a solution. Thought nonetheless that the roundtrip skip could be a starting point worth discussing 😊

About #2390. I'm starting to form the opinion that it does probably not need to be fixed for this use case. Rather, it may even be more straightforward to access once `rt` is separated from `hal`.

[^0]: https://matrix.to/#/!LdaNPfUfvefOLewEIM:matrix.org/$ZE5F8mN_wjxscf-YF6PMT1vtp8URXnhGbDb7BRiGwpY?via=matrix.org&via=tchncs.de&via=mozilla.org
[^1]: https://github.com/esp-rs/esp-hal/blob/3e6b85bf3010f7fac8a104cfca05795559671764/esp-hal/src/lib.rs#L535C1-L563C2
[^2]: https://github.com/esp-rs/esp-hal/pull/1135
[^3]: https://github.com/rust-lang/compiler-team/issues/841
[^4]: https://github.com/esp-rs/esp-hal/blob/3e6b85bf3010f7fac8a104cfca05795559671764/esp-riscv-rt/src/lib.rs#L71C8-L71C30

Dominaezzz, 2025-06-14:
Yes and I agree it's not the most pleasant thing but "awful" is an exaggeration.

The point I'm trying to make here is, `hal_main` isn't hard coded to esp-hal, you can define it yourself, in your own hal.

It also means you can add your own pre-main code, like the stack guard thing esp-hal does.

Dominaezzz, 2025-06-14:
As an aside, I don't think cargo features are a good way to control this behaviour. All it takes is one library in the dependency graph to enable this feature and suddenly your application is doing crazy things you can't explain because the wrong main is being called.

Having a consistent `hal_main` symbol means any conflicts are caught by rustc (as duplicate symbol errors), long before you start losing hairs from debugging. I'd argue this is better UX compared to the ability to skip a couple lines of code.

I'll leave it to the hal maintainers to decide what's sane but in the meantime I'm just letting you know that you're not blocked.

gibbz00, 2025-06-14:
> It also means you can add your own pre-main code, like the stack guard thing esp-hal does.

`__post_init` exists already for this.

https://github.com/esp-rs/esp-hal/blob/3e6b85bf3010f7fac8a104cfca05795559671764/esp-riscv-rt/src/lib.rs#L67

>  All it takes is one library in the dependency graph to enable this feature and suddenly your application is doing crazy things you can't explain because the wrong main is being called.

Uhm, first of, a runtime crate is not supposed to be used by libraries. It exports `global_asm!` 😅 Secondly, there are statics in place to prevent you from doing it in the first place.

https://github.com/esp-rs/esp-hal/blob/3e6b85bf3010f7fac8a104cfca05795559671764/esp-riscv-rt/src/lib.rs#L28-L30

But yes, of course, I myself am not blocked. I could just fork the esp-hal crate, strip out all the non-startup code, rename it to my application, and begin from there. /s Apologies for the sarcasm, but I don't know how else to respond to such a simplistic suggestion. This issue would not have been opened unless I was trying to start a discussion around a solution that could be reused by others with similar interests.

MabezDev, 2025-06-16:
The best way to solve this I think, is to (re-)introduce an `rt` feature on esp-hal. I explain my reasoning more here: https://github.com/esp-rs/esp-hal/issues/2589#issuecomment-2615479990, whilst this mostly focuses on future compatibility, this also applies to your efforts here. You'll be able to disable the `rt` feature on esp-hal, and pull in your own rt crate (or use esp-riscv-rt directly).

There are a few challenges to solve with this approach. One is that without calling `esp_hal::init` there are a tonne of things that won't work correctly, see https://github.com/esp-rs/esp-hal/blob/c29c436be55a4ee2416583be376b2173226bcc3b/esp-hal/src/lib.rs#L605-L685. `init` sets up the `time` module, along with interrupts (required for async). In your case, you'll have to do this manually if you still want to use these APIs.

closed 2025-07-18 by MabezDev