esp-rs/esp-hal #3661: LoadStorePIFAddrError on startup of esp32 no_std Rust program in QEMU

ottobonn, 2025-06-19:
## Bug description

I have an esp32 `no_std` application that I have been developing with QEMU emulation using the official `espressif/idf-rust` Docker image. It was working fine in the QEMU with version 1.74 of the Docker image and Rust, but after upgrading to `espressif/idf-rust:esp32_1.86.0.0` and the latest esp-hal dependencies, I am getting an error when QEMU starts. 

I am currently just running an empty `main.rs` to see what the issue might be, and getting a `LoadStorePIFAddrError`. I haven't been able to find much on this error and would appreciate any leads on what might be going wrong, thanks!

## To Reproduce

This repository contains a minimal reproduction of the error: https://github.com/ottobonn/esp32-qemu-error-repro

## Expected behavior

Previously, this program would boot successfully in QEMU.

## Environment

- Target device: esp32 in QEMU
- Crate name and version:
    ```
    esp-hal = { version = "1.0.0-beta.1", features = ["esp32", "unstable"] }
    esp-backtrace = { version = "0.16.0", features = ["esp32", "panic-handler", "exception-handler", "println"] }
    esp-println = { version = "0.14.0", features = ["esp32"] }
    ```

bugadani, 2025-06-19:
Thanks, the [qemu-side discussion](https://github.com/espressif/qemu/issues/130) made me realize what's going on. We have a bug in the beta.1 linker script, that causes the `__stack_chk_guard` to be placed in an invalid location. Even though you have the stack-smashing protection disabled, the symbol exists and is written during [startup](https://github.com/esp-rs/esp-hal/blob/3cd36062d47b5fdf8ddc6195f9bb1b1e69a83dda/esp-hal/src/soc/esp32/mod.rs#L115-L123).

The problem is observable on physical chips when `-Zstack-protector=all` is enabled, and causes a panic before the chip starts running user code.

This issue has been fixed by #3635 https://github.com/esp-rs/esp-hal/pull/3635/files#diff-617b5c96b6d4364a83c7f79cefdda14114fb33e3a62bc222da8efc832c3ee568R9

closed 2025-06-19 by bugadani

ottobonn, 2025-06-21:
Updating esp-hal to the commit of the fix solved it for me, thanks!