esp-rs/esp-hal #3732: `esp-println` has no output

KortanZ, 2025-07-01:
## Bug description

using esp-generate generate a very simple project, add `esp-println`, then using println! in code. No output in serial port.

Code is like
```rust
#![no_std]
#![no_main]
#![deny(
    clippy::mem_forget,
    reason = "mem::forget is generally not safe to do with esp_hal types, especially those \
    holding buffers for the duration of a data transfer."
)]

use esp_hal::clock::CpuClock;
use esp_hal::main;
use esp_hal::time::{Duration, Instant};
use esp_println::println;

#[panic_handler]
fn panic(_: &core::panic::PanicInfo) -> ! {
    loop {}
}

// This creates a default app-descriptor required by the esp-idf bootloader.
// For more information see: <https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/app_image_format.html#application-description>
esp_bootloader_esp_idf::esp_app_desc!();

#[main]
fn main() -> ! {
    // generator version: 0.4.0

    println!("print test");
    let config = esp_hal::Config::default();
    let _peripherals = esp_hal::init(config);

    loop {
        println!("print test");
        let delay_start = Instant::now();
        while delay_start.elapsed() < Duration::from_millis(500) {}
    }

    // for inspiration have a look at the examples at https://github.com/esp-rs/esp-hal/tree/esp-hal-v1.0.0-beta.1/examples/src/bin
}
```
and output is like
```
~/Documents/code/rust/esp32/tiny-mib master ?8 ❯ cargo run --release                                                                                01:34:52
warning: unused import: `esp_hal::clock::CpuClock`
 --> src/bin/main.rs:9:5
  |
9 | use esp_hal::clock::CpuClock;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

warning: `tiny-mib` (bin "tiny-mib") generated 1 warning (run `cargo fix --bin "tiny-mib"` to apply 1 suggestion)
    Finished `release` profile [optimized + debuginfo] target(s) in 0.04s
     Running `espflash flash --monitor --chip esp32c6 target/riscv32imac-unknown-none-elf/release/tiny-mib`
[2025-07-01T17:34:55Z INFO ] 🚀 A new version of espflash is available: v4.0.0
[2025-07-01T17:34:55Z INFO ] Serial port: '/dev/ttyACM0'
[2025-07-01T17:34:55Z INFO ] Connecting...
[2025-07-01T17:34:55Z INFO ] Using flash stub
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       40:4c:ca:55:e4:2c
App/part. size:    9,296/4,128,768 bytes, 0.23%
[2025-07-01T17:34:57Z INFO ] Segment at address '0x0' has not changed, skipping write
[2025-07-01T17:34:57Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2025-07-01T17:34:57Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2025-07-01T17:34:57Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

I (17) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (18) boot: compile time Jun  7 2023 08:02:08
I (19) boot: chip revision: v0.0
I (23) boot.esp32c6: SPI Speed      : 40MHz
I (27) boot.esp32c6: SPI Mode       : DIO
I (32) boot.esp32c6: SPI Flash Size : 4MB
I (37) boot: Enabling RNG early entropy source...
I (43) boot: Partition Table:
I (46) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (61) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (68) boot:  2 factory          factory app      00 00 00010000 003f0000
I (76) boot: End of partition table
I (80) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=00234h (   564) map
I (88) esp_image: segment 1: paddr=0001025c vaddr=40800000 size=00014h (    20) load
I (97) esp_image: segment 2: paddr=00010278 vaddr=42000278 size=016f4h (  5876) map
I (106) esp_image: segment 3: paddr=00011974 vaddr=40800014 size=00aach (  2732) load
I (114) boot: Loaded app from partition at offset 0x10000
I (119) boot: Disabling RNG early entropy source...
```
## To Reproduce

1. generate project
2. add println!
3. run code with cargo run --release

## Expected behavior

## Environment

- Target device: ESP32C6 
- Crate name and version: 
```
[package]
edition = "2021"
name    = "tiny-mib"
version = "0.1.0"

[[bin]]
name = "tiny-mib"
path = "./src/bin/main.rs"

[dependencies]
esp-bootloader-esp-idf = "0.1.0"
esp-hal                = { version = "=1.0.0-beta.1", features = ["esp32c6"] }

critical-section = "1.2.0"
esp-println = { version = "0.14.0", features = ["esp32c6"] }

[profile.dev]
# Rust debug is too slow.
# For debug builds always builds with some optimization
opt-level = "s"

[profile.release]
codegen-units    = 1     # LLVM can perform better optimizations using a single thread
debug            = 2
debug-assertions = false
incremental      = false
lto              = 'fat'
opt-level        = 's'
overflow-checks  = false
```

KortanZ, 2025-07-02:
Waveshare ESP32-C6-DEV-KIT-N8 development board, it has both UART birdge and jtag. I'm using UART bridge.

bjoernQ, 2025-07-02:
The board apparently comes with an onboard USB-hub which might confuse the auto-detection in esp-println.

You can try this

```diff
- esp-println = { version = "0.14.0", features = ["esp32c6"] }
+ esp-println = { version = "0.14.0", features = ["esp32c6", "uart"], default-features = false }
```

turgu1, 2025-10-07:
Could it be a good idea to add an option to the esp-generate app for this kind of feature?

Jordan-Haidee, 2025-12-06:
This fixed my problem as well — really appreciate it!