esp-rs/esp-hal #3859: Rename `esp-wifi-sys` to `esp-radio-sys`

playfulFence, 2025-07-23:
As an extension of #3815

MabezDev, 2025-07-23:
We might not be able to do this. We got special permission to have increased crate size for esp-wifi-sys so we can ship the blobs with the crate. We'd need to get the same permission for this new crate.

Imo it wouldn't be the end of the world if we had to keep it as esp-wifi-sys, it's not a user facing crate anyways.

playfulFence, 2025-07-23:
I don't mind at all either (especially given the fact this crate exists exclusively for under-the-hood usage), just wanted to explore options to preserve consistency in crate names.
But in this case I'm starting to doubt that the effort is really worth the result

bjoernQ, 2025-07-24:
Ah good point about the size limit exception .... totally forgot about it 🤦‍♂️ 

If someone wants to do it anyways you need to contact the crates.io team.

Another option might be to have an esp-radio-sys which then (depending on the target chip) includes esp-radio-esp32xxx-sys and we split things up. 10mb should be more than enough for one chip

Frostie314159, 2025-08-02:
Another option would be to divide `esp-wifi-sys` into multiple crates, with one for each library.

AnthonyGrondin, 2025-08-11:
All in all, this link and all related links in the organization need to be updated:

https://github.com/esp-rs/esp-wifi-sys/blob/99dd43f9992b276efff3c75467839801e7ea55c7/README.md?plain=1#L5

MabezDev, 2025-09-16:
I think one package per chip makes the most sense here, we can leave `esp-wifi-sys` as is and create `esp-radio-esp32xx-sys` crates for each chip.