esp-rs/esp-hal #3930: RMT driver tracking issue

wisp3rwind, 2025-08-13:
Let's keep track of desirable refactorings and enhancements of the RMT driver. A bunch of these have come up in recent PRs, where it was decided to not address them immediately:

- [ ] Always type-erase the RMT channel number on `Channel`: https://github.com/esp-rs/esp-hal/pull/3505#discussion_r2154655712
  (@wisp3rwind currently working on this)
- [ ] Move channel specification from `cfg_if!` branches to `esp-metadata`: https://github.com/esp-rs/esp-hal/pull/3917#discussion_r2264607504
- [ ] Consider redesigning the extended channel RAM feature to avoid global shared mutable state: https://github.com/esp-rs/esp-hal/pull/3917#discussion_r2264610088
- [ ] Support wrapping rx/tx for all methods
  (@wisp3rwind currently working on this)
- [ ] Support more general data types (iterator, an encoder type similar to IDF)
  (@wisp3rwind currently working on this)
- [ ] Support channel [multi-channel sync tx](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/rmt.html#rmt-multiple-channels-simultaneous-transmission)
- [ ] Implement blocking methods for owned and borrowed channels and redesign how data is passed in: https://github.com/esp-rs/esp-hal/pull/3716#discussion_r2179900769
  (@wisp3rwind currently working on this)
- [ ] Properly tie `Rmt`, `ChannelCreator`, `Channel` lifetimes together.
  (@wisp3rwind currently working on this)

I've marked points that I'm currently working on/have some local prototype for.

kleinesfilmroellchen, 2025-09-22:
Does the point about lifetimes also mean fixing the current problem where you can’t configure the channel more than once, and if you start a transfer and it errors out during the transfer you lose safe access to the channel? Both of those could be fixed by just properly taking the parent structure by mutable reference on creation, which frees them up again after an error or a channel access ended.

kleinesfilmroellchen, 2025-09-22:
For reference, here’s an abomination I had to write (and haven’t tested yet) to emulate a temporarily borrowed transmit channel.

```rust

/// Better interface for borrowing RMT channels temporarily and giving them back after use, able to be reconfigured.
struct BorrowedRmtChannel<'d, const CHANNEL: u8, Int>
where
	Int: ChannelInternal,
{
	// Should usually contain a value, but because of lower-level API problems we have to empty it out once in a while.
	channel: Option<Channel<Blocking, Int>>,
	config:  TxChannelConfig,
	pin:     AnyPin<'d>,
	creator: &'d mut ChannelCreator<Blocking, CHANNEL>,
}

impl<'d> BorrowedRmtChannel<'d, 0, ConstChannelAccess<Tx, 0>> {
	pub fn configure_tx(
		creator: &'d mut ChannelCreator<Blocking, 0>,
		pin: AnyPin<'d>,
		config: TxChannelConfig,
	) -> Result<Self, Error> {
		// SAFETY: We borrow mutably from the channel creator, ensuring that noone else can use it for our lifetime.
		//         The lifetime of the channel is tied to us.
		let actual_channel = unsafe { ChannelCreator::<Blocking, 0>::steal() };
		// SAFETY: Noone except us has a copy of the pin. We only use our second copy in case the channel gets dropped.
		let configured = actual_channel.configure_tx(unsafe { pin.clone_unchecked() }, config)?;
		Ok(Self { channel: Some(configured), config, creator, pin })
	}

	/// Transmit a certain number of loops of the given number.
	/// Due to borrow checker restrictions and the way that channel transmission is implemented in `esp-hal`, this
	/// function consumes the channel and returns it back identically afterwards (instead of taking by mutable
	/// reference). In most situations, you can use this function like `channel = channel.transmit_loops(...)`.
	pub async fn transmit_loops(&mut self, data: &[u32], loops: u16) -> Result<(), Error> {
		debug_assert!(self.channel.is_some());
		let channel = self.channel.take().unwrap();
		let res = channel.transmit_continuously_with_loopcount(loops, data).map_err(|e| {
			// SAFETY: This is a borrow checker workaround. All of this code exits the function. The only problems would
			// happen if this code executed after the map_err below, which it never can.
			*self = Self::configure_tx(
				unsafe { mem::transmute(self.creator as *mut _) },
				unsafe { self.pin.clone_unchecked() },
				self.config,
			)
			.unwrap();
			e
		})?;
		while !res.is_loopcount_interrupt_set() {
			Timer::after(Duration::from_millis(1)).await;
		}
		let channel = res.stop_next().map_err(|(e, channel)| {
			self.channel = Some(channel);
			e
		})?;
		self.channel = Some(channel);
		Ok(())
	}
}

```

bugadani, 2025-09-22:
You can do `Rmt::new(peripherals.RMT.reborrow(), FREQ)` and that way you can reclaim the peripheral singleton once you're done with Rmt. On the other hand, I think this is currently unsound, `ChannelCreator` needs to have a lifetime otherwise it's possible to drop Rmt while channels/channel creator structs are alive, which essentially lets you create any number of copies of the same channel...

kleinesfilmroellchen, 2025-09-22:
> You can do `Rmt::new(peripherals.RMT.reborrow(), FREQ)` and that way you can reclaim the peripheral singleton once you're done with Rmt. 

I need to have two different channels that do their own thing, so I’m not sure this will work, but it’s nice to know that the RMT peripheral can be reborrowed like the GPIOs (and others).

kleinesfilmroellchen, 2025-09-22:
Looking at it again, though, I don’t know what you’re exactly referring to in my code. Since I take `ChannelCreator` by mutable reference, as long as `ChannelCreator` depends on `Rmt` to be alive, things are fine. For me it would also just be adding another lifetime argument in the code above if that were to be changed. But either way, I’d rather want some API upstream that does what I’m hacking in here automatically, and I wanted to give an idea of what kind of API I need in my code. To improve this idea further, it’s probably useful to show the code that uses it as well:

```rust
async fn test_rmt(rmt_peripheral: peripherals::RMT<'_>, mut target: AnyPin<'_>) -> Result<(), esp_hal::rmt::Error> {
        // adjust clock to specific chip, here 80MHz on ESP32-C6
	let mut rmt = Rmt::new(rmt_peripheral, RMT_BASE_CLOCK)?;

        // not important -- calculate the RMT config and pulse data based on some complex logic
	let config = rmt_data_for_steps(RMT_BASE_CLOCK, steps, time).unwrap();

	loop {
		let target = target.reborrow();
		let mut channel = BorrowedRmtChannel::configure_tx(&mut rmt.channel0, target.into(), (&config).into())?;
		channel.transmit_loops(&config.pulses, config.loops).await?;
		Timer::after(Duration::from_millis(1000)).await;
	}
}
```

wisp3rwind, 2025-09-22:
> Does the point about lifetimes also mean fixing the current problem where you can’t configure the channel more than once, 

yes

> and if you start a transfer and it errors out during the initial configuration you lose safe access to the channel?

yes

> Both of those could be fixed by just properly taking the parent structure by mutable reference on creation, which frees them up again after an error or a channel access ended.

not quite, since it is desirable to have an API that takes the channel by value: https://github.com/esp-rs/esp-hal/pull/3716#discussion_r2179900769

---

> You can do `Rmt::new(peripherals.RMT.reborrow(), FREQ)` and that way you can reclaim the peripheral singleton once you're done with Rmt.

That works right now due to the issue you point out below, but it's not a viable solution when using several channels independently.

> On the other hand, I think this is currently unsound, `ChannelCreator` needs to have a lifetime otherwise it's possible to drop Rmt while channels/channel creator structs are alive, which essentially lets you create any number of copies of the same channel...

In fact what you suggest doesn't quite work out, but only by accident since https://github.com/esp-rs/esp-hal/pull/3453: It does compile, but due to tracking some global state it will error out when trying to configure the channel before the first instance is dropped.

---

> Yeah, that’s true. Would be easy enough to add, but doesn’t that mean that you can do unsound things with the upstream API without unsafe since ChannelCreator is not my invention?

Currently, yes, but see the above comment on runtime errors.

---

The (probably) next PR I intend to submit after #4138 will address the above points. It should then be possible to do something along the lines of (note the `.reborrow()` on the `ChannelCreator`)

```rust
async fn test_rmt(rmt_peripheral: peripherals::RMT<'_>, mut target: AnyPin<'_>) -> Result<(), esp_hal::rmt::Error> {
    // adjust clock to specific chip, here 80MHz on ESP32-C6
    let mut rmt = Rmt::new(rmt_peripheral, RMT_BASE_CLOCK)?;

    loop {
        let config = rmt_data_for_steps(RMT_BASE_CLOCK, steps, time).unwrap();
        let channel = rmt.channel0
            .reborrow()
            .configure_tx(target.reborrow(), (&config).into())?;
        let transaction = channel
            .transmit_continuously_with_loopcount(&config.pulses, config.loops)
            .await?;
        while !transaction.is_loopcount_interrupt_set() {}
        Timer::after(Duration::from_millis(1000)).await;
    }
}

```

That PR (mostly ready locally) should enable most use cases that currently don't work, it might not be the most ergonomic for all of those yet.

---

As an aside; does `submit_continuously_with_loopcount` actually work for you? I've been writing a test, and concluded that it's actually mostly broken (esp32c6 being one of the devices where it's reasonably easy to fix, at least).

wisp3rwind, 2025-09-22:
```rust

		// SAFETY: We borrow mutably from the channel creator, ensuring that noone else can use it for our lifetime.
		//         The lifetime of the channel is tied to us.
		let actual_channel = unsafe { ChannelCreator::<Blocking, 0>::steal() };
```

This is enough for memory safety, but there's another problem with current RMT API here: If before the `ChannelCreator::steal()` all `Channel`, `ChannelCreator`s and the `Rmt` have been dropped the peripheral clock will be stopped and re-enabled, which loses the clock configuration.

wisp3rwind, 2025-10-29:
> > As an aside; does `submit_continuously_with_loopcount` actually work for you? I've been writing a test, and concluded that it's actually mostly broken (esp32c6 being one of the devices where it's reasonably easy to fix, at least).
> 
> So I think there’s a bug where it works once and then never again, but this may be related to how well my code is deconfiguring and reconfiguring the peripheral. I have a problem at the moment where there’s several situations in which the second submission will deadlock in this function.

Is that with or without https://github.com/esp-rs/esp-hal/pull/4260? If yes, with which loop mode argument? If not, it think the issue might be that tx actually never stops, even when using a finite loop count. I think there are still issues with continuous tx, but https://github.com/esp-rs/esp-hal/pull/4260 should help. (Note that the migration guide in that PR is a bit incorrect - a fix is pending in https://github.com/esp-rs/esp-hal/pull/4302/commits/13ef4390b9169d2d9cb54eeff373f54f0cb0ff1d

kleinesfilmroellchen, 2025-10-29:
> Is that with or without [#4260](https://github.com/esp-rs/esp-hal/pull/4260)?

1.0.0-rc.0, so without. I am fairly certain I am not currently able to use the main branch on my whole project, as this duplicates some crates in the large dependency tree.

> If not, it think the issue might be that tx actually never stops, even when using a finite loop count. I think there are still issues with continuous tx, but [#4260](https://github.com/esp-rs/esp-hal/pull/4260) should help. (Note that the migration guide in that PR is a bit incorrect - a fix is pending in [13ef439](https://github.com/esp-rs/esp-hal/commit/13ef4390b9169d2d9cb54eeff373f54f0cb0ff1d)

Yes, I think I am observing this occasionally, but so far fixing my own oversights like loop count 0 and empty transmission data has always made it work again.

As an aside, if it helps, my code that makes extensive use of strange RMT configurations (including lots of clock configuration fiddling) is open-source (and not terribly broken anymore): https://git.filmroellchen.eu/filmroellchen/penplot Maybe you can look at the code in https://git.filmroellchen.eu/filmroellchen/penplot/src/branch/main/penplot/cnc/stepping.rs and see how it behaves with any future changes. (For now, as I said, this is difficult for me, and I’m not seeing any major issues with my above-mentioned workaround hacks for the time being -- I’m really just here to suggest API improvements where they would help my use case.)

wisp3rwind, 2025-10-30:
I think what you'd really want here is continuous tx with the simultaneous tx feature (see the TRMs; "multi-channel sync tx" in the list above), which the driver doesn't support.
What might also simplify things is regular tx with wrapping of the hardware tx buffer (supported) and iterator input, which I will soon (after a few other things) submit a PR for.

kleinesfilmroellchen, 2025-10-30:
Just noticed that the simultaneous TX feature exists – would definitely be what I need. It sounds like an entirely separate API (that takes ownership of both channels) would be suitable for this, maybe I can come up with something.

kleinesfilmroellchen, 2025-11-03:
I'm not putting the driver in RAM so this can be related.

kleinesfilmroellchen, 2025-11-03:
The same thing (and unpredictable behavior) happens with `InfiniteWithInterrupt` - except then, if it actually continues to run, it doesn’t stop even when `stop_next` returns.

kleinesfilmroellchen, 2025-11-03:
Pretty sure this is related to me choosing the maximum (or even “just” a large) RMT base clock divider. Take note of the RMT_BASE_CLOCK constant in this almost-minimal reproducer that produces glitched output over 50% of the time:

```rust
#![feature(
    impl_trait_in_assoc_type,
    never_type,
)]
#![no_std]
#![no_main]
#![deny(clippy::mem_forget)]

extern crate alloc;

use anyhow::Result;
use defmt::{debug, trace};
use embassy_executor::Spawner;
use embassy_time::{Duration, Timer};
use esp_hal::clock::CpuClock;
use esp_hal::gpio::{AnyPin, Level};
use esp_hal::interrupt::software::SoftwareInterruptControl;
use esp_hal::rmt::*;
use esp_hal::time::Rate;
use esp_hal::timer::systimer;
use esp_hal::{Blocking, peripherals};
use {esp_alloc as _, esp_backtrace as _, esp_println as _};

esp_bootloader_esp_idf::esp_app_desc!();

const HEAP_SIZE: usize = 1024;

const RMT_BASE_CLOCK: Rate = Rate::from_hz(80_000_000 / 256);

#[esp_rtos::main]
async fn main(spawner: Spawner) -> ! {
    fallible_main(spawner).await.unwrap();
}

async fn fallible_main(_spawner: Spawner) -> Result<!> {
    let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let peripherals: peripherals::Peripherals = esp_hal::init(config);
    esp_alloc::heap_allocator!(size: HEAP_SIZE);
    let systimer = systimer::SystemTimer::new(peripherals.SYSTIMER);
    let sw_int = SoftwareInterruptControl::new(peripherals.SW_INTERRUPT);
    esp_rtos::start(systimer.alarm0, sw_int.software_interrupt0);

    let mut x_step_pin = AnyPin::from(peripherals.GPIO0);
    let mut rmt = Rmt::new(peripherals.RMT, RMT_BASE_CLOCK).unwrap();

    let tx_channel_config = TxChannelConfig::default()
        .with_clk_divider(1)
        .with_carrier_modulation(false)
        .with_idle_output_level(Level::Low)
        .with_idle_output(true);
    const X_RMT_LENGTH: u16 = 289;
    static X_PULSES: [u32; 2] = [
        PulseCode::new(Level::High, X_RMT_LENGTH, Level::Low, X_RMT_LENGTH).0,
        PulseCode::end_marker().0,
    ];

    loop {
        debug!("starting rmt transmission...");
        let x_step_pin = x_step_pin.reborrow();
        let x_channel = rmt
            .channel0
            .reborrow()
            .configure_tx(x_step_pin, tx_channel_config)
            .unwrap();

        // Timer::after_millis(200).await;
        transmit_loops(x_channel, &X_PULSES, 40).await.unwrap();
        // Timer::after_millis(10).await;
        debug!("... done.");
        Timer::after(Duration::from_secs(2)).await;
    }
}

#[unsafe(no_mangle)]
pub extern "Rust" fn _esp_println_timestamp() -> u64 {
    esp_hal::time::Instant::now()
        .duration_since_epoch()
        .as_micros()
}

fn transmit_loops<'ch>(
    channel: Channel<'ch, Blocking, Tx>,
    data: &[u32],
    loops: u16,
) -> impl Future<Output = Result<(), Error>> {
    // Finite and InfiniteWithInterrupts behave the same
    let maybe_res = channel.transmit_continuously(data, LoopMode::Finite(loops));
    trace!("transmission dispatched to hardware");
    async move {
        let res = maybe_res?;
        // does not change the behavior (!)
        // Timer::after_secs(10).await;

        // this code (with or without the if) deadlocks over 50% of the time
        // if loops > 1 {
        trace!("needs to wait for loopcount interrupt...");
        while !res.is_loopcount_interrupt_set() {
            Timer::after(Duration::from_micros(1)).await;
        }
        // }

        trace!("waiting for transmission stop...");
        let _ = res.stop_next().map_err(|(e, _)| e)?;
        trace!("transmission stopped.");
        Ok(())
    }
}

```

Cargo.toml:

```toml
[workspace]
resolver = "3"

[package]
name = "penplot"
edition = "2024"

[[bin]]
path = "penplot/main.rs"
name = "penplot"

[profile.release]
# Symbols are nice and they don't increase the size on Flash
debug = true
opt-level = 3
# maximum optimization with full LTO and minimal parallelism
lto = "fat"
codegen-units = 1
# more safety for the complex arithmetic
overflow-checks = true
debug-assertions = false
incremental = false

[profile.dev]
# Symbols are nice and they don't increase the size on Flash
debug = true
opt-level = "s"

[features]
default = ["esp32c6"]
esp32c6 = [
	"esp-hal/esp32c6",
	"esp-backtrace/esp32c6",
	"esp-println/esp32c6",
	"esp-rtos/esp32c6",
	"esp-bootloader-esp-idf/esp32c6",
]

[dependencies]
esp-bootloader-esp-idf = { version = "0.4", features = ["defmt"] }
esp-hal = { version = "=1.0.0", features = [
	"unstable",
	"defmt",
] }
esp-backtrace = { version = "0.18", features = ["panic-handler", "defmt"] }
esp-alloc = "*"
esp-rtos = { version = "0.2", features = [
	"defmt",
	"embassy",
	"esp-alloc",
] }
esp-println = { version = "0.16", default-features = false, features = [
	"defmt-espflash",
	"timestamp",
	"jtag-serial",
] }
defmt = { version = "*", features = ["alloc"] }

embedded-hal = "1"

embassy-time = { version = "*", features = ["defmt"] }
embassy-executor = { version = "*", features = ["nightly"] }
embassy-sync = { version = "*", features = ["defmt"] }

anyhow = { version = "1", default-features = false }

critical-section = "1"

```

Run with `RUSTFLAGS="-Z stack-protector=all -C link-arg=-Tlinkall.x -C link-arg=-Tdefmt.x -C force-frame-pointers" DEFMT_LOG="debug,esp_rtos=info,penplot=trace" cargo run --package penplot` on latest nightly, ESP32-C6. (Release builds behave the same.)

I will continue my investigation into the clock divider issue, but I hope you can find the bug better with this code at hand.

wisp3rwind, 2025-11-04:
> I don't think any clock-related code has changed between rc.0 and 1.0.0. That said, in a quick test, setting `.with_clk_divider(1)` does break HIL tests in weird ways, whereas `.with_clk_divider(2)` is fine (even if used with double the base frequency, such that the result is the same). The TRM documents some timing constraints for pulse codes, but I don't think they're violated here.

Interestingly, the reset value of the clock divider register field is 2, not 1, which might suggest that 1 is indeed not supported.

wisp3rwind, 2025-11-09:
> I also tried to get the lib from [@kleinesfilmroellchen](https://github.com/kleinesfilmroellchen) to work.
> 
> this patch solved my final issues: [TuxCoder/esp-hal-smartled@bc1d339](https://github.com/TuxCoder/esp-hal-smartled/commit/bc1d3393d1302eba9d1d47a8d5d8e9c2b3827b72)
> 
> I tested with the async api and also with the sync api. And yes, also important to do release build.

Yeah, the RMT can be quite timing-sensitive. I doubt that @kleinesfilmroellchen's issue here and the smartled issue are related: driving WS2812-style LEDs requires much shorter pulses to the point that one must be careful to be able to refill the RMT hardware buffer in time. That's not a concern with with @kleinesfilmroellchen's motor driver, which has much longer pulses and uses continuous tx, which does not require refilling the hardware buffer once the transmission has been started.

In other words, the smartled woes are to be expected; I have some additions to the RMT driver in the pipeline that should help, but ultimately, some care will be required.
The clock divider issue here looks like either an actual bug, or real hardware constraints that the TRMs fail to mention...

EDIT: I'm planning to write down some thoughts on reliably driving LEDs via the RMT at some point, but won't do this right away.

wisp3rwind, 2025-12-09:
> In case I didn’t make this clear: configuring a custom clock source is more complex than just exposing the clock source enum, and this is likely why it’s not exposed. This needs chip-specific register manipulation code (it also depends on whether the chip has the new clock tree config module or not)

The reason it's not exposed is simply that it hasn't been done yet. I've only added the `ClockSource` enum quite recently, as a side effect of moving RMT specification to `esp-metadata`. It makes sense to expose it eventually, but at least the `todo!()` in `ClockSource.freq()` needs to be addressed first.

I'm adding it to the list in the top post.

SergioGasquez, 2026-09-29:
The RMT hangs/corrupted RX with large SCLK dividers (https://github.com/esp-rs/esp-hal/issues/3930#issuecomment-3482788146) is fixed in #6420, which just got merged to `main`. @kleinesfilmroellchen, it would be great if you could confirm it works for your setup on `main`.
 
The factor-2 timing change between 1.0 and 1.1.1 (https://github.com/esp-rs/esp-hal/issues/3930#issuecomment-4525567520) is a separate issue and isn't addressed by #6420.

kleinesfilmroellchen, 2026-10-25:
Apologies, I haven’t been able to get back to it. I will report once I do (this is still in my Github notification list :3); I assume RMT is not close to stabilization?

bugadani, 2026-10-25:
It's next in line, with a very minimal feature set.

kleinesfilmroellchen, 2026-10-25:
Alright, let’s see if I can check it out soon.