esp-rs/esp-hal #3956: ESP32-S3: 80MHz PSRAM config hangs when setting up the allocator

Dominaezzz, 2025-08-19:
## Bug description

```Rust
#[main]
async fn main(spawner: Spawner) {
    let peripherals = esp_hal::init(
        esp_hal::Config::default()
            .with_cpu_clock(CpuClock::max())
            .with_psram(PsramConfig {
                ram_frequency: SpiRamFreq::Freq80m,
                ..Default::default()
            })
    );

    let (_, size) = psram_raw_parts(&peripherals.PSRAM);
    println!("PSRAM size = {}", size);

    esp_alloc::heap_allocator!(#[link_section = ".dram2_uninit"] size: 64000);
    esp_alloc::heap_allocator!(size: 130 * 1024);
    esp_alloc::psram_allocator!(peripherals.PSRAM, esp_hal::psram);

    println!("This text doesn't show up!");
}
```

Same as reported [here](https://github.com/esp-rs/esp-hal/issues/1566#issuecomment-2376070258).

## To Reproduce

I have not tried this on `main` yet.

1. ...
2. ...

## Expected behavior

Don't hang

## Environment

- Target device: ESP32-S3 
- Crate name and version: esp-hal 1.0.0-rc.0

bjoernQ, 2025-08-20:
Ah maybe the octal-PSRAM@80MHz problem is something else ... I am not sure it ever worked (checked a few versions) at all

So in your case you are not using octal and you see it printing the PSRAM size but it dies when creating the allocator, right?

Dominaezzz, 2025-08-20:
I'm using one of these. https://docs.m5stack.com/en/unit/Unit-CAMS3%205MP

I'm indeed trying to use Octal PSRAM @ 80MHz. (Sorry, I meant to say Octal in the issue desc but forgot while I was crafting the code snippet)

> you see it printing the PSRAM size but it dies when creating the allocator, right?

Yes! It also dies (or halts rather) when I try to write to the raw pointer. It appears any memory access to PSRAM causes it to hang.

closed 2025-08-22 by MabezDev