esp-rs/esp-hal #4004: Exception 'Load access fault'

Szybet, 2025-08-28:
## Bug description

Every few days, while booting my program, this happens just after turning on BLE:
```
5.848220 [INFO ] esp-radio configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: true, ampdu_tx_enable: true, amsdu_tx_enable: false, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 } (esp_radio esp-radio/src/lib.rs:278)
5.903834 [ERROR] panicked at 'Exception 'Load access fault' mepc=420ef8f0, mtval=ffd0000f
TrapFrame { ra: 1108277650, t0: 202, t1: 1108942680, t2: 32, t3: 0, t4: 0, t5: 573046784, t6: 14680064, a0: 8, a1: 1082462208, a2: 2, a3: 1082195356, a4: 1, a5: 1611325440, a6: 1082215564, a7: 0 }' (esp_hal src/exception_handler/mod.rs:46)
5.904106 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904129 [ERROR] ====================== PANIC ====================== (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904151 [ERROR] panicked at src/main.rs:43:5 (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904244 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904265 [ERROR] Backtrace: (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904287 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904343 [ERROR] 0x420fdc4e (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904378 [ERROR] 0x42112d46 (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904413 [ERROR] 0x40800494 (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904448 [ERROR] 0x40800348 (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904483 [ERROR] 0x420ef940 (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904518 [ERROR] 0x420ef862 (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904553 [ERROR] 0x420fdc22 (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904588 [ERROR] 0x420ef688 (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904626 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904647 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
5.904669 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
Firmware exited with: Application exit (Aborted)
Core 0
    Frame 0: syscall_readonly @ 0x42188f64 inline
       .cargo/registry/src/index.crates.io-1949cf8c6b5b557f/semihosting-0.1.20/src/sys/arm_compat/syscall/riscv.rs:35:9
    Frame 1: sys_exit_extended @ 0x0000000042188f40 inline
       .cargo/registry/src/index.crates.io-1949cf8c6b5b557f/semihosting-0.1.20/src/sys/arm_compat/mod.rs:191:9
    Frame 2: exit @ 0x0000000042188f40 inline
       .cargo/registry/src/index.crates.io-1949cf8c6b5b557f/semihosting-0.1.20/src/sys/arm_compat/mod.rs:160:5
    Frame 3: exit @ 0x0000000042188f40
       .cargo/registry/src/index.crates.io-1949cf8c6b5b557f/semihosting-0.1.20/src/process.rs:57:5
    Frame 4: abort @ 0x42188f9c
       .cargo/registry/src/index.crates.io-1949cf8c6b5b557f/semihosting-0.1.20/src/process.rs:69:5
    Frame 5: abort @ 0x4210ec4e
       .cargo/git/checkouts/esp-hal-f15fbdeaa22f7411/efb4ea4/esp-backtrace/src/lib.rs:249:13
    Frame 6: panic_handler @ 0x4210ebfe
       .cargo/git/checkouts/esp-hal-f15fbdeaa22f7411/efb4ea4/esp-backtrace/src/lib.rs:120:5
    Frame 7: panic_fmt @ 0x421093ce
       .rustup/toolchains/nightly-2025-05-13-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/panicking.rs:75:14
    Frame 8: panic @ 0x420fdc50
       init-esp32-c6/src/main.rs:43:5
    Frame 9: panic @ 0x42112d48 inline
       .cargo/registry/src/index.crates.io-1949cf8c6b5b557f/defmt-1.0.1/src/export/mod.rs:133:14
    Frame 10: ExceptionHandler @ 0x0000000042112d42
       .cargo/git/checkouts/esp-hal-f15fbdeaa22f7411/efb4ea4/esp-hal/src/exception_handler/mod.rs:0
    Frame 11: panic @ 0x42112d48 inline
       .cargo/registry/src/index.crates.io-1949cf8c6b5b557f/defmt-1.0.1/src/export/mod.rs:133:14
    Frame 12: ExceptionHandler @ 0x0000000042112d42
       .cargo/git/checkouts/esp-hal-f15fbdeaa22f7411/efb4ea4/esp-hal/src/exception_handler/mod.rs:0
```

## To Reproduce

Good question, happens every few days, once, then it's gone for a while. Haven't found a way to replicate it or appear it more often

## Environment

- Target device: esp32c6
- Crate name and version: esp-hal at commit `085beda621fb39a97b81fcb43b11b9b832d89278` (So like 3 days old)

MabezDev, 2025-08-29:
What's on line `init-esp32-c6/src/main.rs:43:5`?

Szybet, 2025-09-02:
Well it's not reliable to reproduce in my program (Maybe would be if probe-rs attach wouldn't crash the device sometimes...), so I doubt trying to create a reproductible example is possible

Anyway, another weird BLE error, this is a new one, I never encrountered it:
```
5.850725 [INFO ] esp-radio configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: true, ampdu_tx_enable: true, amsdu_tx_enable: false, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 } (esp_radio esp-radio/src/lib.rs:279)
5.915551 [ERROR] panicked at 'ASSERT r_ble_hci_trans_hs_cmd_tx:214 0 0' (esp_radio src/ble/npl.rs:423)
5.915730 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
5.915752 [ERROR] ====================== PANIC ====================== (esp_backtrace esp-backtrace/src/lib.rs:70)
5.915774 [ERROR] panicked at src/main.rs:44:5 (esp_backtrace esp-backtrace/src/lib.rs:70)
5.915870 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
5.915891 [ERROR] Backtrace: (esp_backtrace esp-backtrace/src/lib.rs:70)
5.915913 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
5.915966 [ERROR] 0x420ffd16 (esp_backtrace esp-backtrace/src/lib.rs:70)
5.916007 [ERROR] 0x4211a01c (esp_backtrace esp-backtrace/src/lib.rs:70)
5.916045 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
5.916066 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
5.916088 [ERROR]  (esp_backtrace esp-backtrace/src/lib.rs:70)
Firmware exited with: Application exit (Aborted)
ERROR probe_rs_debug::debug_info: Other("UNWIND: Tried to unwind `RegisterRule` at CFA = None.")
Core 0
    Frame 0: syscall_readonly @ 0x4218c354 inline
       ~/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/semihosting-0.1.20/src/sys/arm_compat/syscall/riscv.rs:35:9
    Frame 1: sys_exit_extended @ 0x000000004218c330 inline
       ~/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/semihosting-0.1.20/src/sys/arm_compat/mod.rs:191:9
    Frame 2: exit @ 0x000000004218c330 inline
       ~/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/semihosting-0.1.20/src/sys/arm_compat/mod.rs:160:5
    Frame 3: exit @ 0x000000004218c330
       ~/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/semihosting-0.1.20/src/process.rs:57:5
    Frame 4: abort @ 0x4218c38c
       ~/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/semihosting-0.1.20/src/process.rs:69:5
    Frame 5: abort @ 0x42110cea
       ~/.cargo/git/checkouts/esp-hal-f15fbdeaa22f7411/677a933/esp-backtrace/src/lib.rs:249:13
    Frame 6: panic_handler @ 0x42110c9a
       ~/.cargo/git/checkouts/esp-hal-f15fbdeaa22f7411/677a933/esp-backtrace/src/lib.rs:120:5
    Frame 7: panic_fmt @ 0x4210b472
       ~/.rustup/toolchains/nightly-2025-05-13-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/panicking.rs:75:14
    Frame 8: panic @ 0x420ffd18
       coolpathinit/init-esp32-c6/src/main.rs:44:5
    Frame 9: panic @ 0x4211a01e inline
       ~/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/defmt-1.0.1/src/export/mod.rs:133:14
    Frame 10: osi_assert @ 0x000000004211a018
       ~/.cargo/git/checkouts/esp-hal-f15fbdeaa22f7411/677a933/esp-radio/src/ble/npl.rs:423:9
    Frame 11: <unknown function @ 0x4219aa6a> @ 0x4219aa6a

    Frame 12: <unknown function @ 0x4219aa6a> @ 0x4219aa6a

    Frame 13: <unknown function @ 0x4219aa6a> @ 0x4219aa6a

    Frame 14: <unknown function @ 0x0000fffe> @ 0x0000fffe

    Frame 15: <unknown function @ 0x4085401e> @ 0x4085401e

    Frame 16: {async_fn#0} @ 0x42158d96 inline
       coolpathlogic/src/lib.rs
    Frame 17: poll<logic::__led_task_task::{async_fn_env#0}> : ERROR: UNWIND: Tried to unwind `RegisterRule` at CFA = None. @ 0x0000000042158ce0
       ~/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/embassy-executor-0.7.0/src/raw/mod.rs:214:15
```