esp-rs/esp-hal #4137: Merge esp-hal-embassy & esp-preempt into `esp-runtime`

MabezDev, 2025-09-18:
After some discussion with @bugadani, we realized these crates both do very similar things, though esp-preempt can do more (threads, etc).

In theory, we can merge these two crates and turn it into a `esp-runtime` crate (in reality its more likely we'll implement what need for embassy in esp-preempt, and rename).

Merging the two has a few benefits:

* Tighter integration between anything using threads, like esp-radio (executor can be thread aware etc)
  * We can get a better view of the system as a whole, and create better idling and power saving features
* A single timer queue (we currently have an impl in each crate)
* Saves a timer for users wanting to use esp-radio + embassy, as the timer queue can be shared between the two

bugadani, 2025-09-18:
No, the executor will still be an opt-in feature, and will need to be started by the user.

finalyards, 2025-09-19:
This sounds awesome. Currently, `esp-hal-embassy` is keeping me from upgrading to `embassy-executor` 0.9.1; it works with 0.7.0.

pebender, 2025-09-29:
The merge of esp-hal-embassy into esp-preempt is broken on the esp32c6. I will open new issue for it.

PR #4206 (2025-09-30): Rename esp-preempt to esp-rtos
Closes #4137
closed 2025-09-30 by bugadani