esp-rs/esp-hal #4149: Cannot get defmt to work

Verequies, 2025-09-19:
I am trying to compile the latest embassy-hello-world-defmt example for the ESP32C6.

If I am successful in compilation I always get this error with defmt-print, espflash and probe-rs:
```failed to demangle defmt symbol `__global_pointer$`: expected value at line 1 column 1```

If I do compile an older example and try to attempt to get defmt to work, I always get no response.
The probe is the standard ESP JTAG probe.

Surely I must be missing something here or is it that I cannot get logs out?
defmt works with my RP2350 based projects using Embassy just fine.

bugadani, 2025-09-19:
Try [this](https://github.com/esp-rs/esp-generate/pull/226) please

Verequies, 2025-09-19:
@bugadani Thank you, I was able to get the example working with espflash. probe-rs however for some reason programs quite a bit slower and does not show any messages still. Any idea what is going on there?

bugadani, 2025-09-19:
I have no idea, it works for me.

Verequies, 2025-09-19:
Ah yes, makes sense. Swapped that over and now its working. Thank you again!

Interestingly when programming with probe-rs and a log file is specified using --log-file, the programming speed is around 30KB/s. When no log file is specified I was seeing upwards of 80 KB/s.

MabezDev, 2025-09-19:
When using a log file it logs _all_ log levels to file, including trace, this slows things down considerably.

Glad you got things working now :)

closed 2025-09-19 by MabezDev

Verequies, 2025-09-19:
@MabezDev, that does make sense, but I was sending the log to `/dev/null` and it was still the same. For the RP2350, the same drops it down to about 100 KB/s from 110KB/s which is understandable with the overhead. Given this, I don't get why the speed is dropped way down to 30 KB/s for the ESP32C6.