esp-rs/esp-hal #4156: Refactor I2S master module

vaknin, 2025-09-20:
### Problem

esp-hal/src/i2s/master.rs has grown to roughly 2.7 kLOC. The single file now mixes configuration objects, clock math, hardware/register plumbing, blocking driver logic, async helpers, and per-chip special cases, which makes the driver hard to navigate, reason about, and review.

### Proposed Solution

Reorganize i2s::master as a directory module so each concern lives in its own file while preserving the existing public path, esp_hal::i2s::master. The async submodule keeps the current asynch spelling because async is reserved and downstream code already depends on that name.

 esp-hal/src/i2s/master/
  ├── mod.rs       // existing docs + re-exports (replaces master.rs)
  ├── config.rs    // Config/UnitConfig/ConfigError and validation helpers (~460 lines)
  ├── clock.rs     // I2S clock divider math (~100 lines)
  ├── hardware.rs  // “private” hardware abstraction, register access, per-SoC logic (~1,200 lines)
  ├── driver.rs    // Blocking driver types (I2s/I2sTx/I2sRx), DMA plumbing (~500 lines)
  └── asynch.rs    // Async DMA wrappers and futures (~220 lines)

### Benefits

  - Maintainability: Focused modules map to clear responsibilities, easing reviews.
  - Readability: Developers can jump straight to config, clocks, driver, async, etc.
  - Compilation ergonomics: Smaller translation units reduce incremental build churn.
  - Extensibility: Leaves headroom for new features without re-inflating a single file.
  - No API churn: master/mod.rs re-exports the same items, so downstream crates continue to compile unchanged.

Dominaezzz, 2025-09-20:
Almost every other driver is organized in the same way. Having this one be different is probably not ideal.

Some structs in `esp-hal/src/i2s/master.rs` could probably be moved to `esp-hal/src/i2s/mod.rs`, if it's going to be shared with the slave driver.

There's also another refactor brewing [here](https://github.com/esp-rs/esp-hal/pull/3639) and the file might become easier to parse after that.
The author hasn't been able to drag it to the finish line yet but I'd prefer if the judgement about how big the file is happened after that PR (or a similar one achieving the same refactor) lands.

vaknin, 2025-09-22:
> My monitor can display ~50 lines of code comfortably, and rust-analyzer/IDEs already makes navigating the codebase pretty easy.

I mean, we could technically fit all of esp_hal in a single file, but that's probably not the way to go.

If you don't find this issue necessary, feel free to close it.
The reason I created this specific issue is because it felt like the first step in enhancing the i²s code in esp-hal (a thing that I would like to assist with and I find interesting), and that would make it more manageable.

I was sarcastic with this, should've been clearer:
> I guess that means we should refactor everything!

That was just uncalled for though, and this attitude will repel new contributors from joining:
> Besides the seemingly LLM-generated list of advantages

redengin, 2026-01-17:
Piling on to raise awareness of this.

Maintainability: Focused modules map to clear responsibilities, easing reviews.
The current design of I2S is horrendous - unnecessarily abstracts I2S control making it only usable for very specific use-cases.

For me I want to implement  something like [ESP32FMRadio](https://github.com/Alexxdal/ESP32FMRadio) - but the current I2S public interfaces make that impossible.