esp-rs/esp-hal #4193: Embassy Timer not working correctly with SPI Slave on ESP32-C6

Verequies, 2025-09-26:
There seems to be an issue with Embassy and using the timers to await while using the SPI Slave on the ESP32-C6.

Looking at the below code in `host_task`, specifically the while `!transfer.is_done()` loop.
If you only `yield_now`, or use `Timer::after`, they do not work and no other tasks or even the main loop is given time to work. I have even noticed weird events such as the transfer loop waiting till the `Timer::after` completes in the main loop. Eventually this results in a complete freeze of the ESP32-C6, using within a few seconds.

Surprisingly, I have found that if you use a `defmt::flush()` before a `yield_now().await` or `Timer::after`, then everything stabilises and the other tasks run fine and I the RGB LED onboard gets the updates correctly.

How `defmt::flush()` is doing this, I am not sure, but clearly there is an issue here given that is makes a completely non functional program, functional without crashes.

Is there any alternative to `defmt::flush()` I can use in the mean time, till a fix is properly implemented?

```rust
#![no_std]
#![no_main]
#![feature(impl_trait_in_assoc_type)]
#![deny(
    clippy::mem_forget,
    reason = "mem::forget is generally not safe to do with esp_hal types, especially those \
    holding buffers for the duration of a data transfer."
)]

use defmt::info;
use defmt_rtt as _;
use embassy_executor::Spawner;
use embassy_futures::yield_now;
use embassy_sync::blocking_mutex::raw::CriticalSectionRawMutex;
use embassy_sync::signal::Signal;
use embassy_time::Timer;
use esp_backtrace as _;
use esp_hal::clock::CpuClock;
use esp_hal::rmt::Rmt;
use esp_hal::spi::slave::dma::SpiDma;
use esp_hal::time::Rate;
use esp_hal::timer::systimer::SystemTimer;
use esp_hal::Blocking;
use esp_hal::{
    dma::{DmaRxBuf, DmaTxBuf},
    dma_buffers,
    spi::{slave::Spi, Mode},
};
use led::{ChannelOrder, LedPayload, Rgb, RmtWs2812};

extern crate alloc;

esp_bootloader_esp_idf::esp_app_desc!();

pub static CHA_WIFI_RGB: Signal<CriticalSectionRawMutex, LedPayload> = Signal::new();

#[esp_hal_embassy::main]
async fn main(spawner: Spawner) -> ! {
    let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let peripherals = esp_hal::init(config);

    esp_alloc::heap_allocator!(size: 64 * 1024);

    let timer0 = SystemTimer::new(peripherals.SYSTIMER);
    esp_hal_embassy::init(timer0.alarm0);    

    // RMT
    let rmt_clock = Rate::from_mhz(80);
    let rmt = Rmt::new(peripherals.RMT, rmt_clock).unwrap().into_async();

    // SYSLED
    let rmt_divider = 2;
    let ws2812 = RmtWs2812::new(
        rmt.channel0,
        rmt_clock,
        rmt_divider,
        peripherals.GPIO8,
        ChannelOrder::RGB,
        1,
    );

    // SPI pins
    let miso = peripherals.GPIO0;
    let sclk = peripherals.GPIO1;
    let mosi = peripherals.GPIO2;
    let cs = peripherals.GPIO3;

    // DMA buffers
    let (rx_buffer, rx_descriptors, tx_buffer, tx_descriptors) = dma_buffers!(40);
    let dma_rx_buf = DmaRxBuf::new(rx_descriptors, rx_buffer).unwrap();
    let dma_tx_buf = DmaTxBuf::new(tx_descriptors, tx_buffer).unwrap();

    // SPI slave instance (Blocking mode)
    let host_comms = Spi::new(peripherals.SPI2, Mode::_0)
        .with_sck(sclk)
        .with_mosi(mosi)
        .with_miso(miso)
        .with_cs(cs)
        .with_dma(peripherals.DMA_CH0);

    // Spawn SPI task
    spawner.spawn(host_task(host_comms, dma_rx_buf, dma_tx_buf).expect("Host Task Spawning Failed"));

    // Spawn SYSLED task
    spawner.spawn(sysled_task(ws2812).expect("SYSLED Task Spawning Failed"));
    
    // Main loop
    info!("Booted!");
    loop {
        info!("Still alive!");
        Timer::after_secs(1).await;
    }
}

#[embassy_executor::task]
async fn sysled_task(mut sysled: RmtWs2812) -> ! {
    let mut rgb: LedPayload;

    loop {
        if CHA_WIFI_RGB.signaled() {
            rgb = CHA_WIFI_RGB.wait().await;
            sysled.write(&[rgb.rgb], rgb.brightness as u16).await;
        }
        Timer::after_micros(10).await;
    }
}

#[embassy_executor::task]
async fn host_task(
    mut host_comms: SpiDma<'static, Blocking>,
    mut dma_rx_buf: DmaRxBuf,
    mut dma_tx_buf: DmaTxBuf,
) -> ! {
    loop {
        // Prepare buffers
        dma_tx_buf.as_mut_slice().fill(0x00);
        dma_rx_buf.as_mut_slice().fill(0x00);
        let tx_data = b"WACK";
        dma_tx_buf.as_mut_slice()[..tx_data.len()].copy_from_slice(tx_data);

        let transfer = host_comms
            .transfer(dma_rx_buf.len(), dma_rx_buf, dma_tx_buf.len(), dma_tx_buf)
            .unwrap();

        while !transfer.is_done() {
            defmt::flush();
            yield_now().await;
        }

        (host_comms, (dma_rx_buf, dma_tx_buf)) = transfer.wait();
        
        let mut rx = [0u8; 4];
        let mut offset = 0;
        for chunk in dma_rx_buf.received_data() {
            let len = chunk.len().min(rx.len() - offset);
            rx[offset..offset + len].copy_from_slice(&chunk[..len]);
            offset += len;
            if offset >= rx.len() {
                break;
            }
        }

        match core::str::from_utf8(&rx) {
            Ok(txt) => defmt::info!("Received: {}", txt),
            Err(_) => defmt::warn!("Non-UTF8 frame: {:?}", &rx),
        }

        CHA_WIFI_RGB.signal(LedPayload { rgb: Rgb::new(rx[0], rx[1], rx[2]), brightness: rx[3] });
    }
}

```

bugadani, 2025-09-26:
My advice is: run your blocking code in the main function without an embassy executor. Run the rest of your async code in InterruptExecutor(s). That way the async code can run wherever it needs to run, at a higher priority than the blocking transfer. SPI slave is not in our focus, and unless this is a wider-spread issue, a fix may come later than you'd like.

That being said, your sysled_task is a lot more heavy than you'd like it to be. The 10us wait pretty much says it's looping as often as it can, or not much more spaced out than that. Just waiting for the signal would be equivalent, since you only write to the RMT when the signal is signaled. While embassy is fair, this is something you should look at.

Verequies, 2025-09-26:
Thanks for the suggestion. I had a play around with the `esp_hal::delay::Delay` and it does sort of work initially, but then ultimately results in a complete system freeze. The SYSLED updates were either slowed down significantly or missed entirely. 

I did manage to get it stable without `defmt::flush()` and crashes (for now) by using a larger timer of 500 micros instead of 10 micros in the sysled_task, however that did result in missed updates. I also tried with release and debug builds too - just to confirm it wasn't anything to do with compilation flags/options.

So the question still remains to why `defmt::flush()` helps and makes it all run without issue.

bugadani, 2025-09-27:
Just a complete wild guess, but what happens if you set ESP_HAL_EMBASSY_CONFIG_LOW_POWER_WAIT to false? It would also be interesting to test what happens with https://github.com/esp-rs/esp-hal/pull/4172 (as generally the thread-mode executor will be somewhat slower) but that's a bigger task, I'll try to look into that on monday.

bugadani, 2025-09-30:
Okay I've tried your code, but I had to trim out the smartled stuff because I don't know what crate/module `led` is supposed to be. I'm also using rtt-target, for time reasons (I used esp-generate to create my project).

Unfortunately I can't reproduce your issue like this, for me the `Still alive` prints come on time as expected. If you can share your complete project for me to try, that would be great.

Verequies, 2025-09-30:
@bugadani The `led` crate is an internal crate and currently I do not want to open it up. But it does not rely on the `smartled` crate at all.

The issue can be replicated with this [respository](https://github.com/Verequies/low_power_timer_issue). I pushed my changes to `esp-hal` to my [fork](https://github.com/Verequies/esp-hal). Mainly changes to make it compatible with with upstream Embassy so that I can use it in my multi-arch project.

`defmt::flush()` has been commented out and `ESP_HAL_EMBASSY_CONFIG_LOW_POWER_WAIT` set to `true`, so that it should compile straight away to a state that does not work. Disabling `ESP_HAL_EMBASSY_CONFIG_LOW_POWER_WAIT` or uncommenting `defmt::flush()` will make it work again.

Verequies, 2025-09-30:
I assume you mean the rename from `__pender` to `__espender`?
The rename is required due to duplications with Embassy. The project would not compile with out that change.
If there is another way around this I could try without the rename.
```
warning: Linking globals named '__pender': symbol multiply defined!

error: failed to load bitcode of module "esp_hal_embassy-aa325fc479634cde.b0tte3rmqr9dh4j1akobucenc.14cfgxr.rcgu.o": 
```

bugadani, 2025-09-30:
`embassy-executor = { version = "0.7.0", features = ["nightly", "executor-thread", "defmt", "arch-riscv32"] }`

Do not enable the `arch-riscv32` feature.

Verequies, 2025-09-30:
Yep, that seems to compile now. Strange, I could have sworn that embassy-executor required a platform specific feature flag in order to get running. I definitely require `arch-cortex-m` for the RP2350 side of things.

Anyhow, so far after undoing the `__pender` rename, it seems to be running fine without issues. Guess it was really just something I was doing.

Thanks for catching that!
Perhaps a warning should be said somewhere to make sure not to have `arch-riscv32` enabled for `embassy-executor`?