esp-rs/esp-hal #4210: The merge of esp-hal-embassy into esp-preempt is broken on the esp32c6.

pebender, 2025-09-29:
## Bug description

The merge of esp-hal-embassy into esp-preempt has broken the embassy executor on esp32c6. Futures are being starved.

## To Reproduce

Complile and run the modified version of examples/wifi/embassy_dhcp found at <https://github.com/pebender/esp-hal/commit/0dbb4358e2e73d10fa286d8c93d928b632b56861> using

cargo run --release --features esp32c6 --target riscv32imac-unknown-none-elf

with the latest stable rust compiler: 1.90.0 (1159e78c4 2025-09-14).

The only change to the example is that it prints out the value of Instant in seconds every second rather than fetching a URL. Instead of printing out every second, it sometimes misses multiple seconds. Here is an example output (with ssid and bssid replaced with ***)

start connection task
Device capabilities: Ok(EnumSet(Client | AccessPoint | ApSta))
Starting wifi
Wifi started!
Scan
AccessPointInfo { ssid: "***", bssid: [***], channel: 7, secondary_channel: Above, signal_strength: -32, auth_method: Some(Wpa2Personal), country: None }
AccessPointInfo { ssid: "***", bssid: [***], channel: 7, secondary_channel: Above, signal_strength: -59, auth_method: Some(Wpa2Personal), country: None }
AccessPointInfo { ssid: "***", bssid: [***], channel: 7, secondary_channel: Above, signal_strength: -64, auth_method: Some(Wpa2Personal), country: None }
AccessPointInfo { ssid: "***", bssid: [***], channel: 9, secondary_channel: None, signal_strength: -87, auth_method: Some(Wpa2Personal), country: Some(Country([85, 83])) }
AccessPointInfo { ssid: "***", bssid: [***], channel: 9, secondary_channel: None, signal_strength: -87, auth_method: Some(Wpa2Personal), country: Some(Country([85, 83])) }
About to connect...
Wifi connected!
Waiting to get IP address...
Got IP: 192.168.1.132/24
connecting...
connected!
18
19
24
25
26
27
28
30
35
38
44
48
54
55
56
58

## Expected behavior

It should print out a value once a second.

## Environment

The target is:

Chip type:         esp32c6 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi 6, BT 5
MAC address:       f0:f5:bd:02:b3:c0

Security Information:
=====================
Flags: 0x00000000 (0)
Key Purposes: [0, 0, 0, 0, 0, 0, 12]
Chip ID: 13
API Version: 0
Secure Boot: Disabled
Flash Encryption: Disabled
SPI Boot Crypt Count (SPI_BOOT_CRYPT_CNT): 0x0

bugadani, 2025-09-29:
You can set the priority of the main thread above the priority of the wifi thread (which is 29)

pebender, 2025-10-01:
I finally got around to testing it and the problem I was seeing has been resolved. Thanks for fixing this so quickly.

PR #4212 (2025-09-30): Fix embassy timer queue
Closes #4210

This PR fixes an issue where adding a new embassy task to the timer queue incorrectly overwrote the next wakeup time. This resulted in tasks not getting readied on time. The solution is to correctly track the minimum of the stored wakeup time and the new deadline.